coulisse/spiderweb #1: Possible to add search function ?

efiten, 2020-09-09:
Is it possible to ad a search/filter option for only 1 call? We we can check when a call is last seen in the database?

coulisse, 2020-09-09:
Hello, I'm glad to know that you are using spiderweb.
Concerning your request, yes, this request is in the TODO list; I've just to have the time to do that :-)

It will be the next thing that I'll do.

Best regards
Corrado

efiten, 2020-09-09:
Yes, I was looking for a decent web interface for a long time, because the
dxspider interface isn't that good.
And installation was easy, with some knowledge of linux & git.

So yes, you're doing a great job !

coulisse, 2020-09-14:
Hello
I started to work on it: I think I'll end the development within 2 weeks.

Meanwhile could we can connect our two DX Cluster nodes?

My info are:
    node-call = iu1bow-2
    address: dxcluster.iu1bow.it
    tcp/port: 7300
    Software: DX SPIDER
    Web address: https://dxcluster.iu1bow.it

73 DE IU1BOW - Corrado

coulisse, 2020-09-20:
Solved with 1.2 version

closed 2020-09-20 by coulisse

efiten, 2020-09-20:
An what's the best way to update to the lastest version ?

A git pull won't start the update (I'm not that familiar with git): 

`root@debian-2gb-nbg1-1:~/spiderweb# git pull
remote: Enumerating objects: 84, done.
remote: Counting objects: 100% (84/84), done.
remote: Compressing objects: 100% (37/37), done.
remote: Total 61 (delta 28), reused 55 (delta 23), pack-reused 0
Unpacking objects: 100% (61/61), done.
From https://github.com/coulisse/spiderweb
   6cf676d..1987f34  master     -> origin/master
Updating 6cf676d..1987f34
error: Your local changes to the following files would be overwritten by merge:
        static/sitemap.xml
        templates/index.html
        templates/plots.html
Please commit your changes or stash them before you merge.
Aborting
`

efiten, 2020-09-20:
FYI, i've added you as a spider node, my info :

Call : ON0NOL-9
Connect to : nolcluster.on8ar.eu Port 7300

coulisse, 2020-09-20:
Hello 

> An what's the best way to update to the lastest version ?
> 
> A git pull won't start the update (I'm not that familiar with git):

If you are not familiar to git I think the best method is to make a backup copy of your folder, then delete it and download the latest version using clone command:

**git clone https://github.com/coulisse/spiderweb.git**

Then copy the file **config.json** from you backup copy, with your previous settings.

Regards
Corrado