coulisse/spiderweb #2: Improving the spots filter

s50u, 2021-01-02:
Ciao

Would you please consider improving the DXDe and DX spots filtering. The idea is to download Jim's AD1C free special version of CTY.DAT once a week from http://www.country-files.com/category/big-cty/ and used it for better spots filtering. The cty.dat format file structure is Column | Lenght | Description. Seems 2 letter continental abbreviation is a good starting point.

1 | 26 | Country Name
27 | 5 | CQ Zone
32 | 5 | ITU Zone
37 | 5 | 2-letter continent abbreviation
42 | 9 | Latitude in degrees, + for North
51 | 10 | Longitude in degrees, + for West
61 | 9 | Local time offset from GMT
70 | 6 | Primary DXCC Prefix (A “*” preceding this prefix indicates that the  country is on the DARC WAEDC list, and counts in CQ-sponsored contests,  but not ARRL-sponsored contests).

More on file format at http://www.country-files.com/cty-dat-format/

Grazie, Danilo S50U

coulisse, 2021-01-04:
Hi Danilo, thanks for your suggestions.

Now I'm working on a release in branch named "bs5_py3". When I'll release (I think whithin jenuary 2021)  I think I could work on this reuqest

coulisse, 2021-12-05:
Hi Danilo,
a long time has passed.

However now I would start to work on this issue.

I found many type of files:

https://www.country-files.com/big-cty/  --> with CSV or DAT file
https://www.country-files.com/cty/wpxloc.raw 
https://www.country-files.com/cty/cty.dat

Which one do you recommend? (and why).

Thanks
Corrado

s50u, 2021-12-05:
> Which one do you recommend? (and why)

I was in touch with Jim and his suggestion would be as follows:

If you are going to download once/week, then you don't need bigcty.  This is actually the best one to use:

https://www.country-files.com/cty/cty_wt_mod.dat

It has a complete listing of prefixes and callsigns.
It has better geographical coordinates for many prefixes (i.e. W7, LU4X, etc.) as compared to the country's coordinates.
It contains the ADIF country code.

It is very similar to cty.dat.  It uses '#' for comments. If you find it easier to parse a .CSV file, then this one is probably a good choice.  It  has the same data as cty_wt_mod.dat:

https://www.country-files.com/cty/cty.csv

Some of the lines are very long. The fields are:

Display Prefix, Name, ADIF code, CONT, CQ Zone, ITU Zone, LAT (+ = north), LONG (+ = west), prefixes

coulisse, 2021-12-07:
Hello Danilo,
I'm working on it.
It's more complex due to manage many exceptions...

For example in the folloing case, "GB3LER" is Scotland or Shetland Islands? There is a rule to manage this case?

___
```
Scotland:                 14:  27:  EU:   56.82:     4.18:     0.0:  GM:
    2A,2M,GM,GS,MA,MM,MS,=GB0DBS,=GB0GDS,=GB0SJR,=GB1002ZE,=GB100BCG,=GB2ELH,=GB2JCM,=GB2OWM,=GB2ZE,
    =GB3ANG,=GB3LER,=GB3LER/B,=GB4GDS,=GB4LER,=GB60ANT,=GM3WUX/2ZE,=GM4UBJ/2ZE,=GM8IEM/2ZE;

Shetland Islands:         14:  27:  EU:   60.50:     1.50:     0.0:  *GM/s:
    =2M0BDR,=2M0BDT,=2M0CPN,=2M0GFC,=2M0SEG,=2M0SPX,=2M0ZET,=GB2ELH,=GB3LER,=GB3LER/B,=GB4LER,=GM0AVR,
    =GM0CXQ,=GM0EKM,=GM0GFL,=GM0ILB,=GM0JDB,=GM1FGN,=GM1KKI,=GM1ZNR,=GM3WHT,=GM3ZET,=GM4IPK,=GM4JPI,
    =GM4LBE,=GM4LER,=GM4PXG,=GM4SLV,=GM4SSA,=GM4WXQ,=GM4ZHL,=GM6RQW,=GM6YQA,=GM7AFE,=GM7GWW,=GM8LNH,
    =GM8MMA,=GM8YEC,=GS3ZET,=MM0LSM,=MM0NQY,=MM0VIK,=MM0XAU,=MM0ZAL,=MM0ZCG,=MM0ZRC,=MM1FEO,=MM1FJM,
    =MM3VQO,=MM5PSL,=MM5YLO,=MM6BDU,=MM6BZQ,=MM6IKB,=MM6IMB,=MM6MFA,=MM6PTE,=MM6SJK,=MM6YLO,=MM6ZBG,
    =MM6ZDW,=MM7CGR,=MM8A,=MS0ZCG,=MS0ZET;_
```
___

Thanks in advance,
Corrado

coulisse, 2021-12-07:
Hello
i the meanwhile I developed  a test api, you cand use in this way:
https://test.iu1bow.it/callsign?c=IU1BOW

could you make any test and tell me if the resoults are good?

Thanks
Corrado

s50u, 2021-12-07:
> For example in the folloing case, "GB3LER" is Scotland or Shetland Islands? There is a rule to manage this case?

In this case it doesn't matter if you filter by CONT = Continent

s50u, 2021-12-07:
> i the meanwhile I developed a test api, you cand use in this way:
> https://test.iu1bow.it/callsign?c=IU1BOW

I checked few latest dxpeditions and all seems fine with the latest CTY-3129 version dated on 30.11.2021. You can now extend the application filtering not only by CONTINENT but by CQ ZONE as well.  👍

coulisse, 2021-12-07:
Thanks for you check; now I'll implement in my spiderweb.

Just another issue: I need to have to country ISO code (IT for Italy, DE for Germany... ), in order to show the right flag.
do you have an idea, how can I get it? Or the only way is to transcode the descriptions?

Thx
Corrado

s50u, 2021-12-07:
># ADIF 499
>Slovenia:                 15:  28:  EU:   46.00:   -14.00:    -1.0:  S5:
>    S5;

Is there any ADIF country code to ISO 3166 converter on the market? :-)

https://en.wikipedia.org/wiki/ISO_3166-1

CU Danilo

coulisse, 2021-12-11:
> 

Hello regarding the CQ filtering the only way is basing it on mysql db information, recorded by dxspider cluster.
In order to do this, in a correct way you have to update every week manually the CTY.dat file. See procedure at the following link
[https://www.country-files.com/dx-cluster/dx-spider/](https://www.country-files.com/dx-cluster/dx-spider/)

So I will to implement a filter for CQ / ITU zone ad I'll put an option on cfg/config.json for activate this filter (default non active)

Bye 
Corrado

s50u, 2021-12-11:
Cool.
This is easily achieved not manually but with the dxspider crontab routine look at mine below. It is downloaded and installed with the procedure starting every Friday at 2:00 AM automatically to be ready at 2:45 AM for coming contest weekend. It won't download unless the file is newer.

######### PETEK 2 AM - PRIDOBI ZADNJO CTY.DAT IN WPXLOC.RAW TER JIH UVOZI
0  2 * * 5 spawn("cd /spider/local_data && wget -qN https://www.country-files.com/cty/cty.dat")
15 2 * * 5 spawn("cd /spider/local_data && wget -qN https://www.country-files.com/cty/wpxloc.raw")
30 2 * * 5 spawn("cd /spider/local_data && /spider/perl/create_prefix.pl")
45 2 * * 5 run_cmd("load/prefixes")

73 Danilo S50U

coulisse, 2021-12-12:
Hi Danilo 
I added the cq filter.
You can test it on [test.iu1bow.it](https://test.iu1bow.it)

I put the code on v2.2 branch (for the moment this is just  a beta version)

Let me know your comments.
Best regards
Corrado

s50u, 2021-12-12:
Hi

When searching from ZONE CQ 15 for CQ 03 selecting only MODE CW I got spots including FT8! Otherwise seems filter by ZONE is working correctly. Perhaps it would be better to have a selectable filter for ZONE 1-40.

73 Danilo

coulisse, 2021-12-19:
Regarding the Mode, I opened a new issue.
Regarding Zone filter I made some changes.

Could you check it on  [test.iu1bow.it](https://test.iu1bow.it)   ?

Thanks,
Corrado

s50u, 2021-12-19:
Seems all ok, nice selection filter! Maybe you devide filters with

De/Dx

**OR**

CQ De ZONE
CQ Dx ZONE

Danilo

coulisse, 2021-12-23:
Hi Danilo
could you write me on my personal mail corrado.gerbaldo@gmail.com because I have a question about dxspider crontab.

thanks
Corrado

coulisse, 2022-01-13:
Closed with 2.2 version

closed 2022-01-13 by coulisse