coulisse/spiderweb #7: Plot & Stats generates an exception

manolab, 2021-11-05:
Hi!

The plots& stats pages throws an exception...

2021-11-05 18:32:45,656 [ERROR] (app.py) Exception on /plots.html [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 2051, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1501, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1499, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1485, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "/opt/spiderweb-master/webapp.py", line 206, in plots
    whoj=who_is_connected()
  File "/opt/spiderweb-master/webapp.py", line 170, in who_is_connected
    response=who(host_port[0],host_port[1],cfg['mycallsign'])
  File "/opt/spiderweb-master/lib/dxtelnet.py", line 61, in who
    return parse_who(res)
  File "/opt/spiderweb-master/lib/dxtelnet.py", line 14, in parse_who
    lines = lines.splitlines()
AttributeError: 'int' object has no attribute 'splitlines'

Can you advice on how to resolve the issue?
Thanks!

manolab, 2021-11-05:
And it seems that spots are not updating on the page...

coulisse, 2021-11-06:
Hello
from the browser do you see plots, or have you also an error on the browser?

manolab, 2021-11-06:
I see an error (500).

Inviato da iPhone

coulisse, 2021-11-06:
Could you run python3 webapp.py from a console and give me the log?

manolab, 2021-11-06:
2021-11-06 22:43:53,870 [INFO] (webapp.py) Start
2021-11-06 22:43:53,872 [INFO] (qry.py) [Errno 2] No such file or directory: '../cfg/config.json'
2021-11-06 22:43:53,873 [INFO] (qry.py) trying with other path...
2021-11-06 22:43:53,873 [INFO] (qry.py) config file loaded
2021-11-06 22:43:53,878 [INFO] (qry.py) db connection pool created
2021-11-06 22:43:53,878 [INFO] (adxo.py) connection to: http://dxcal.kj4z.com/dxcal
2021-11-06 22:43:54,191 [INFO] (adxo.py) number ADXO events: 31
 * Serving Flask app 'webapp' (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
2021-11-06 22:43:54,201 [WARNING] (_internal.py)  * Running on all addresses.
   WARNING: This is a development server. Do not use it in a production deployment.
2021-11-06 22:43:54,201 [INFO] (_internal.py)  * Running on http://46.101.248.139:5000/ (Press CTRL+C to quit)
2021-11-06 22:45:36,749 [INFO] (_internal.py) 212.43.113.113 - - [06/Nov/2021 22:45:36] "GET / HTTP/1.1" 200 -
2021-11-06 22:45:36,817 [INFO] (_internal.py) 212.43.113.113 - - [06/Nov/2021 22:45:36] "GET /static/favicon.png HTTP/1.1" 304 -
2021-11-06 22:45:36,819 [INFO] (_internal.py) 212.43.113.113 - - [06/Nov/2021 22:45:36] "GET /static/js/min/load-sw.min.js HTTP/1.1" 304 -
2021-11-06 22:45:36,820 [INFO] (_internal.py) 212.43.113.113 - - [06/Nov/2021 22:45:36] "GET /static/js/min/index_inline.min.js HTTP/1.1" 304 -
2021-11-06 22:45:36,821 [INFO] (_internal.py) 212.43.113.113 - - [06/Nov/2021 22:45:36] "GET /static/js/min/application.min.js HTTP/1.1" 304 -
2021-11-06 22:45:36,824 [INFO] (_internal.py) 212.43.113.113 - - [06/Nov/2021 22:45:36] "GET /static/images/background.webp HTTP/1.1" 304 -
2021-11-06 22:45:36,868 [INFO] (_internal.py) 212.43.113.113 - - [06/Nov/2021 22:45:36] "GET /static/css/min/style.min.css HTTP/1.1" 200 -
2021-11-06 22:45:42,640 [ERROR] (app.py) Exception on /plots.html [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 2051, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1501, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1499, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1485, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
  File "webapp.py", line 206, in plots
    whoj=who_is_connected()
  File "webapp.py", line 170, in who_is_connected
    response=who(host_port[0],host_port[1],cfg['mycallsign'])
  File "/opt/spiderweb-master/lib/dxtelnet.py", line 61, in who
    return parse_who(res)
  File "/opt/spiderweb-master/lib/dxtelnet.py", line 14, in parse_who
    lines = lines.splitlines()
AttributeError: 'int' object has no attribute 'splitlines'
2021-11-06 22:45:42,643 [INFO] (_internal.py) 212.43.113.113 - - [06/Nov/2021 22:45:42] "GET /plots.html HTTP/1.1" 500 -

coulisse, 2021-11-08:
mmm it's seems depend on the in struction lines = lines.splitlines(), present in /opt/spiderweb-master/lib/dxtelnet.py

Could you enable "DEBUG" mode, changing in file webapp.py the line n. 18 and setting it to true? 
i.e.
app.config["DEBUG"] = True

After that could you run again the application and give me the log?

Thansk
Corrado

manolab, 2021-11-09:
Attached the output in the browser, here’s the output on the console:

 

# python3 webapp.py 

2021-11-09 07:38:58,140 [INFO] (webapp.py) Start

2021-11-09 07:38:58,142 [INFO] (qry.py) [Errno 2] No such file or directory: '../cfg/config.json'

2021-11-09 07:38:58,143 [INFO] (qry.py) trying with other path...

2021-11-09 07:38:58,143 [INFO] (qry.py) config file loaded

2021-11-09 07:38:58,148 [INFO] (qry.py) db connection pool created

2021-11-09 07:38:58,149 [INFO] (adxo.py) connection to: http://dxcal.kj4z.com/dxcal

2021-11-09 07:38:58,459 [INFO] (adxo.py) number ADXO events: 30

* Serving Flask app 'webapp' (lazy loading)

* Environment: production

   WARNING: This is a development server. Do not use it in a production deployment.

   Use a production WSGI server instead.

* Debug mode: on

2021-11-09 07:38:58,478 [WARNING] (_internal.py)  * Running on all addresses.

   WARNING: This is a development server. Do not use it in a production deployment.

2021-11-09 07:38:58,479 [INFO] (_internal.py)  * Running on http://46.101.248.139:5000/ (Press CTRL+C to quit)

2021-11-09 07:38:58,481 [INFO] (_internal.py)  * Restarting with stat

2021-11-09 07:38:59,035 [INFO] (webapp.py) Start

2021-11-09 07:38:59,038 [INFO] (qry.py) [Errno 2] No such file or directory: '../cfg/config.json'

2021-11-09 07:38:59,038 [INFO] (qry.py) trying with other path...

2021-11-09 07:38:59,038 [INFO] (qry.py) config file loaded

2021-11-09 07:38:59,044 [INFO] (qry.py) db connection pool created

2021-11-09 07:38:59,044 [INFO] (adxo.py) connection to: http://dxcal.kj4z.com/dxcal

2021-11-09 07:38:59,347 [INFO] (adxo.py) number ADXO events: 30

2021-11-09 07:38:59,359 [WARNING] (_internal.py)  * Debugger is active!

2021-11-09 07:38:59,359 [INFO] (_internal.py)  * Debugger PIN: 121-723-523

2021-11-09 07:40:27,037 [INFO] (_internal.py) 195.32.84.39 - - [09/Nov/2021 07:40:27] "GET / HTTP/1.1" 200 -

2021-11-09 07:40:27,184 [INFO] (_internal.py) 195.32.84.39 - - [09/Nov/2021 07:40:27] "GET /static/css/min/style.min.css HTTP/1.1" 200 -

2021-11-09 07:40:27,228 [INFO] (_internal.py) 195.32.84.39 - - [09/Nov/2021 07:40:27] "GET /static/js/min/index_inline.min.js HTTP/1.1" 200 -

2021-11-09 07:40:27,229 [INFO] (_internal.py) 195.32.84.39 - - [09/Nov/2021 07:40:27] "GET /static/js/min/load-sw.min.js HTTP/1.1" 200 -

2021-11-09 07:40:27,281 [INFO] (_internal.py) 195.32.84.39 - - [09/Nov/2021 07:40:27] "GET /static/js/min/application.min.js HTTP/1.1" 200 -

2021-11-09 07:40:27,392 [INFO] (_internal.py) 195.32.84.39 - - [09/Nov/2021 07:40:27] "GET /static/favicon.png HTTP/1.1" 200 -

2021-11-09 07:40:27,419 [INFO] (_internal.py) 195.32.84.39 - - [09/Nov/2021 07:40:27] "GET /static/images/background.webp HTTP/1.1" 200 -

2021-11-09 07:40:27,733 [INFO] (_internal.py) 195.32.84.39 - - [09/Nov/2021 07:40:27] "GET /static/images/icons/icon-apple.png HTTP/1.1" 200 -

2021-11-09 07:40:33,883 [INFO] (_internal.py) 195.32.84.39 - - [09/Nov/2021 07:40:33] "GET /plots.html HTTP/1.1" 500 -

Traceback (most recent call last):

  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 2069, in __call__

    return self.wsgi_app(environ, start_response)

  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 2054, in wsgi_app

    response = self.handle_exception(e)

 File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 2051, in wsgi_app

    response = self.full_dispatch_request()

  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1501, in full_dispatch_request

    rv = self.handle_user_exception(e)

  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1499, in full_dispatch_request

    rv = self.dispatch_request()

  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1485, in dispatch_request

    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)

  File "/opt/spiderweb-master/webapp.py", line 206, in plots

    whoj=who_is_connected()

  File "/opt/spiderweb-master/webapp.py", line 170, in who_is_connected

    response=who(host_port[0],host_port[1],cfg['mycallsign'])

  File "/opt/spiderweb-master/lib/dxtelnet.py", line 61, in who

    return parse_who(res)

  File "/opt/spiderweb-master/lib/dxtelnet.py", line 14, in parse_who

    lines = lines.splitlines()

AttributeError: 'int' object has no attribute 'splitlines'

2021-11-09 07:40:33,943 [INFO] (_internal.py) 195.32.84.39 - - [09/Nov/2021 07:40:33] "GET /plots.html?__debugger__=yes&cmd=resource&f=style.css HTTP/1.1" 200 -

2021-11-09 07:40:33,975 [INFO] (_internal.py) 195.32.84.39 - - [09/Nov/2021 07:40:33] "GET /plots.html?__debugger__=yes&cmd=resource&f=debugger.js HTTP/1.1" 200 -

2021-11-09 07:40:34,054 [INFO] (_internal.py) 195.32.84.39 - - [09/Nov/2021 07:40:34] "GET /plots.html?__debugger__=yes&cmd=resource&f=console.png HTTP/1.1" 200 -

2021-11-09 07:40:34,056 [INFO] (_internal.py) 195.32.84.39 - - [09/Nov/2021 07:40:34] "GET /plots.html?__debugger__=yes&cmd=resource&f=ubuntu.ttf HTTP/1.1" 200 -

2021-11-09 07:40:34,068 [INFO] (_internal.py) 195.32.84.39 - - [09/Nov/2021 07:40:34] "GET /plots.html?__debugger__=yes&cmd=resource&f=console.png HTTP/1.1" 200 -

^C^CException ignored in: <module 'threading' from '/usr/lib/python3.7/threading.py'>

Traceback (most recent call last):

  File "/usr/lib/python3.7/threading.py", line 1281, in _shutdown

    t.join()

  File "/usr/lib/python3.7/threading.py", line 1032, in join

    self._wait_for_tstate_lock()

  File "/usr/lib/python3.7/threading.py", line 1048, in _wait_for_tstate_lock

    elif lock.acquire(block, timeout):

KeyboardInterrupt

 

Thanks,

Normando

coulisse, 2021-11-11:
Hello, 
sorry but I dont find the problem.
Could you replace dextelnet.py  you can see in the link below and give me the log?

[dxtelnet.py.txt](https://github.com/coulisse/spiderweb/files/7519532/dxtelnet.py.txt)

Rename it removing .txt extension

Thanks.
Corrado

manolab, 2021-11-11:
2021-11-11 10:20:56,057 [INFO] (webapp.py) Start

2021-11-11 10:20:56,060 [INFO] (qry.py) [Errno 2] No such file or directory: '../cfg/config.json'

2021-11-11 10:20:56,060 [INFO] (qry.py) trying with other path...

2021-11-11 10:20:56,061 [INFO] (qry.py) config file loaded

2021-11-11 10:20:56,066 [INFO] (qry.py) db connection pool created

2021-11-11 10:20:56,067 [INFO] (adxo.py) connection to: http://dxcal.kj4z.com/dxcal

2021-11-11 10:20:56,560 [INFO] (adxo.py) number ADXO events: 30

* Serving Flask app 'webapp' (lazy loading)

* Environment: production

   WARNING: This is a development server. Do not use it in a production deployment.

   Use a production WSGI server instead.

* Debug mode: on

2021-11-11 10:20:56,585 [WARNING] (_internal.py)  * Running on all addresses.

   WARNING: This is a development server. Do not use it in a production deployment.

2021-11-11 10:20:56,586 [INFO] (_internal.py)  * Running on http://46.101.248.139:5000/ (Press CTRL+C to quit)

2021-11-11 10:20:56,588 [INFO] (_internal.py)  * Restarting with stat

2021-11-11 10:20:57,292 [INFO] (webapp.py) Start

2021-11-11 10:20:57,294 [INFO] (qry.py) [Errno 2] No such file or directory: '../cfg/config.json'

2021-11-11 10:20:57,294 [INFO] (qry.py) trying with other path...

2021-11-11 10:20:57,295 [INFO] (qry.py) config file loaded

2021-11-11 10:20:57,300 [INFO] (qry.py) db connection pool created

2021-11-11 10:20:57,301 [INFO] (adxo.py) connection to: http://dxcal.kj4z.com/dxcal

2021-11-11 10:20:57,605 [INFO] (adxo.py) number ADXO events: 30

2021-11-11 10:20:57,618 [WARNING] (_internal.py)  * Debugger is active!

2021-11-11 10:20:57,618 [INFO] (_internal.py)  * Debugger PIN: 121-723-523

2021-11-11 10:21:07,593 [INFO] (_internal.py) 195.32.84.39 - - [11/Nov/2021 10:21:07] "GET / HTTP/1.1" 200 -

2021-11-11 10:21:07,706 [INFO] (_internal.py) 195.32.84.39 - - [11/Nov/2021 10:21:07] "GET /static/css/min/style.min.css HTTP/1.1" 200 -

2021-11-11 10:21:07,741 [INFO] (_internal.py) 195.32.84.39 - - [11/Nov/2021 10:21:07] "GET /static/js/min/load-sw.min.js HTTP/1.1" 200 -

2021-11-11 10:21:07,746 [INFO] (_internal.py) 195.32.84.39 - - [11/Nov/2021 10:21:07] "GET /static/js/min/index_inline.min.js HTTP/1.1" 200 -

2021-11-11 10:21:07,747 [INFO] (_internal.py) 195.32.84.39 - - [11/Nov/2021 10:21:07] "GET /static/js/min/application.min.js HTTP/1.1" 200 -

2021-11-11 10:21:07,873 [INFO] (_internal.py) 195.32.84.39 - - [11/Nov/2021 10:21:07] "GET /static/favicon.png HTTP/1.1" 200 -

2021-11-11 10:21:07,896 [INFO] (_internal.py) 195.32.84.39 - - [11/Nov/2021 10:21:07] "GET /static/images/background.webp HTTP/1.1" 200 -

2021-11-11 10:21:08,217 [INFO] (_internal.py) 195.32.84.39 - - [11/Nov/2021 10:21:08] "GET /static/images/icons/icon-apple.png HTTP/1.1" 200 -

==============================

type: 

<class 'int'>

..............................

raw: 

0

..............................

data: 

ERROR on object lines

2021-11-11 10:21:14,368 [INFO] (_internal.py) 195.32.84.39 - - [11/Nov/2021 10:21:14] "GET /plots.html HTTP/1.1" 500 -

Traceback (most recent call last):

  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 2069, in __call__

    return self.wsgi_app(environ, start_response)

  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 2054, in wsgi_app

    response = self.handle_exception(e)

  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 2051, in wsgi_app

    response = self.full_dispatch_request()

  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1501, in full_dispatch_request

    rv = self.handle_user_exception(e)

  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1499, in full_dispatch_request

    rv = self.dispatch_request()

  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 1485, in dispatch_request

    return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)

  File "/opt/spiderweb-master/webapp.py", line 206, in plots

    whoj=who_is_connected()

  File "/opt/spiderweb-master/webapp.py", line 170, in who_is_connected

    response=who(host_port[0],host_port[1],cfg['mycallsign'])

  File "/opt/spiderweb-master/lib/dxtelnet.py", line 73, in who

    return parse_who(res)

  File "/opt/spiderweb-master/lib/dxtelnet.py", line 26, in parse_who

    lines = lines.splitlines()

AttributeError: 'int' object has no attribute 'splitlines'

2021-11-11 10:21:14,458 [INFO] (_internal.py) 195.32.84.39 - - [11/Nov/2021 10:21:14] "GET /plots.html?__debugger__=yes&cmd=resource&f=style.css HTTP/1.1" 200 -

2021-11-11 10:21:14,459 [INFO] (_internal.py) 195.32.84.39 - - [11/Nov/2021 10:21:14] "GET /plots.html?__debugger__=yes&cmd=resource&f=debugger.js HTTP/1.1" 200 -

2021-11-11 10:21:14,538 [INFO] (_internal.py) 195.32.84.39 - - [11/Nov/2021 10:21:14] "GET /plots.html?__debugger__=yes&cmd=resource&f=console.png HTTP/1.1" 200 -

2021-11-11 10:21:14,540 [INFO] (_internal.py) 195.32.84.39 - - [11/Nov/2021 10:21:14] "GET /plots.html?__debugger__=yes&cmd=resource&f=ubuntu.ttf HTTP/1.1" 200 -

2021-11-11 10:21:14,565 [INFO] (_internal.py) 195.32.84.39 - - [11/Nov/2021 10:21:14] "GET /plots.html?__debugger__=yes&cmd=resource&f=console.png HTTP/1.1" 200 -

^C^CException ignored in: <module 'threading' from '/usr/lib/python3.7/threading.py'>

Traceback (most recent call last):

  File "/usr/lib/python3.7/threading.py", line 1281, in _shutdown

    t.join()

  File "/usr/lib/python3.7/threading.py", line 1032, in join

    self._wait_for_tstate_lock()

  File "/usr/lib/python3.7/threading.py", line 1048, in _wait_for_tstate_lock

    elif lock.acquire(block, timeout):

KeyboardInterrupt

coulisse, 2021-11-11:
It seems that this module couldn't connect to your telnet host.
Pelase check:
1) Is your dxspider running?
2) Have you correctly setted the 	"telnet" item in your config.json?

Thanks

manolab, 2021-11-11:
I had an error on “MyCallsign”, I included -6 thinking it was the “cluster’s callsign with SSID”.

 

Thanks for the support!

coulisse, 2021-11-11:
Good!
However I have to fix my code, in order to catch this issue and report it in log.

73!
Corrado

closed 2021-11-11 by coulisse