coulisse/spiderweb #8: World clock

s50u, 2021-12-12:
Hi

It would be very useful to see real utc time clock with current date. Maybe it would fit just before "callsign" search field.

73 Danilo S50U

coulisse, 2021-12-19:
Hello I made this change on 2.2 beta.

See it on [test.iu1bow.it](https://test.iu1bow.it)

Could you check it?

Thanks,
Corrado

s50u, 2021-12-19:
Hi

I would prefer live clock in COOKIE format like "Current time:  **Sunday, 19-Dec-2021 12:05:49 UTC**"
See: https://www.utctime.net/

Ciao Danilo

coulisse, 2021-12-23:
Ciao Danilo,
I preferr only hh:mm, because is more lean and less distracting, but is just a personal opinion.

However if you want, you could change  the javascript  static/js/clock.js and working on the string "time" in order to obtain your preferred formad.

Keep in mind that you have to minify the javascript executing the script in scripts/build.sh

Ciao
Corrado

s50u, 2021-12-26:
ok I would prefer live clock from the pic.

73 Danilo

[image]

coulisse, 2022-01-13:
Closed with 2.2 version

closed 2022-01-13 by coulisse