coulisse/spiderweb #9: Mode FT8 when selected CW

coulisse, 2021-12-19:
When searching from ZONE CQ 15 for CQ 03 selecting only MODE CW I got spots including FT8!

s50u, 2021-12-19:
Maybe FT8 spots could be filtered out by "comments" field. If string <ft8> in comments field then pass/reject. Not an ideal solution but no better idea come to my mind at the moment.

73 Danilo S50U

coulisse, 2021-12-23:
At the moment if have in mode.json max and min freq for each mode  / band.

See [https://github.com/coulisse/spiderweb/blob/2.2/cfg/modes.json](https://github.com/coulisse/spiderweb/blob/2.2/cfg/modes.json)

What do you think if simply add the JT9/JT65/FT8 standard  freq in digi section and remove them from phone/cw section?

If yes could you help me editing this file?

Thanks
Corrado

s50u, 2021-12-23:
How about using dxspider's band structure? Here is a copy of the 160m band from the bands file.

> 		  '160m' => bless( { band => [ 1800, 2000 ], 
>							 cw => [ 1800, 1838 ], 
>							 rtty => [ 1838, 1841 ],
>							 data => [ 1838, 1843],
>							 ssb => [ 1831, 2000] 
>						   }, 'Bands'),

coulisse, 2021-12-28:
Hello
i tried to parse the file and i generated modes.json you can find in attachments.
However the result is not so good. In fact some frequencies mode are "overlapped"

For example:

        '15m' => bless( { band => [ 21000, 21450 ],
                            cw => [ 21000, 21150 ],
                            data => [ 21070, 21119 ],
                             rtty => [ 21070, 21119 ],
                             ssb => [ 21151, 21450]
                           }, 'Bands'),

FT8 freq is "21074" and in bands.pl is in range cw (21000,21150) and in data (21070,21119)

[modes.zip](https://github.com/coulisse/spiderweb/files/7783665/modes.zip)

coulisse, 2022-01-03:
Hello
i made some changes on original modes.json (only for HF at the moment).

you can see it in action on [test.iu1bow.it](https://test.iu1bow.it)

Could you make some test, please?
In attachment you can find the new version of modes.json
[modes.zip](https://github.com/coulisse/spiderweb/files/7801402/modes.zip)

s50u, 2022-01-03:
HNY 2022!
Found first freq overlap on 18.100 with many FT8 spots which are sjown so IF conditions is also needed in your code if FT8 is in comment field. And/or simply to consider band plan change for CW on 17m is from 18068.0 - 18099.9 as example.

{
  "modes": [
    {
      "id": "cw",
      "freq": [
	    {
          "min": 18068.0,
          "max": 18099.9
        },

s50u, 2022-01-04:
Same with 1.840, 3.570, 7.040kHz and so on ...

coulisse, 2022-01-10:
Hi Danilo
I pushed a new commit on branch 2.2 changing the band limits (with  ".9" on the max limit) as you suggested. 

At the moment I don't consider the comments (I hope this will not necessary).

However you can test it at [test.iu1bow.it](https://test.iu1bow.it)

Waiting for your feedback
best regards
Corrado

s50u, 2022-01-12:
Hi
Seems band limits did the trick fine.

73 Danilo

coulisse, 2022-01-13:
Closed with 2.2 version

closed 2022-01-13 by coulisse