coulisse/spiderweb #20: Installation issues

videobuff, 2022-09-18:
Using Centos 7.
dxspider up and running well with the name pa0esh-3. I have followed the instructions to set up the webinterface, however i get stuck here:

```
[sysop@pa0esh spiderweb]$ pip install -r requirements.txt
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Defaulting to user installation because normal site-packages is not writeable
ERROR: Could not find a version that satisfies the requirement attrs==22.1.0 (from -r requirements.txt (line 1)) (from versions: 15.0.0a1, 15.0.0, 15.1.0, 15.2.0, 16.0.0, 16.1.0, 16.2.0, 16.3.0, 17.1.0, 17.2.0, 17.3.0, 17.4.0, 18.1.0, 18.2.0, 19.1.0, 19.2.0, 19.3.0, 20.1.0, 20.2.0, 20.3.0, 21.1.0, 21.2.0, 21.3.0, 21.4.0)
ERROR: No matching distribution found for attrs==22.1.0 (from -r requirements.txt (line 1))
[sysop@pa0esh spiderweb]$ 
```
I have both python 2.7 as well as python 3.6.8 in stalled. 
Centos 7 seems to revert to python 2.7. Using alternatives i managed to get the deault to 3.6.8:
sudo alternatives --config python and then choose 3.6.8

[sysop@pa0esh spiderweb]$ python --version
Python 3.6.8
[sysop@pa0esh spiderweb]$ 

It still gives me the same error.

Any help is realyy appreciated,

Erik, PA0ESH

coulisse, 2022-09-19:
Hello Erik,
it seems the problem is that you are using python 2.7 instead 3.6.8.
I don't know how use python 3.6.8 as default.
You could try to install using pip3 that will refer to python 3.6.8

`pip3 install -r requirements.txt`

Waiting for your reply.
Kind regards.
Corrado - IU1BOW

videobuff, 2022-09-19:
Thanks very much Corrado. I got that part solved. However. the installation stops at line 2. 
certifi @ file:///opt/conda/conda-bld/certifi_1655968806487/work/certifi
The directory cannot be found.  It is also not in the git package....
Sorry to bother you but i hope you can help me further.

Erik

videobuff, 2022-09-19:
Dear Corrado,

That worked partially. I had to install certifi. BUt then again i got stuck so i deleted that rule for the time being and tried again. Now my pyton verson is too old , it needs to be 3.10, working on that right now.
I'll get back if i get stuck again...
73's
Erik

videobuff, 2022-09-19:
Hi Corrado,

After a couple of hours i give up. On Centos it is almost impossible to sortout the required python version and pip version etc. I installed python3.10, and used an alias to make it default. But then pip complains. I sorted that out and then flask complains

[sysop@pa0esh spiderweb]$ sudo pip install Flask-minify
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Collecting Flask-minify
  Downloading Flask-Minify-0.39.tar.gz (8.8 kB)
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-KMyhqE/flask-minify/setup.py'"'"'; __file__='"'"'/tmp/pip-install-KMyhqE/flask-minify/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-fvab2G
         cwd: /tmp/pip-install-KMyhqE/flask-minify/
    Complete output (6 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-KMyhqE/flask-minify/setup.py", line 72
        *supported_python_classifiers,
        ^
    SyntaxError: invalid syntax
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
[

Last question ? What OS you are using for dxspider and spiderweb ?

73's

Erik

coulisse, 2022-09-20:
Hello I'm using **ubuntu 18.04-LTS** deployed on a VM Standard B1ls in Azure: it cost less than a raspberry :-)

However you are using pip instead of pip3 command: could you try with pip3 ?
Onother way is to make a simbolic link like:
`sudo ln -s /usr/bin/python3.8 /usr/bin/python`

You can also use [conda](https://docs.conda.io/en/latest/) in order to define a separate python environment (but I'm not an expert)

73's
Corrado

coulisse, 2022-09-26:
Hello Erik,
have you any news?

Thanks
Corrado

closed 2022-10-05 by coulisse