coulisse/spiderweb #22: propagation_heatmaps.sh fails with 'Passing a Normalize instance simultaneously with vmin/vmax is not supported.'

minitajfun, 2022-09-25:
**Description**
When trying to generate heatmaps with the script, it fails after plotting message with error below
`
Traceback (most recent call last):
  File "/spiderweb/scripts/../lib/propagation_heatmaps.py", line 142, in <module>
    im = plt.imshow(np.array(number_ar), cmap='YlOrRd', interpolation='none', norm=LogNorm(vmin=10, vmax=35),vmin=max(np.array(number_ar).min(), LOGMIN))
  File "/usr/local/lib/python3.9/dist-packages/matplotlib/_api/deprecation.py", line 454, in wrapper
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/matplotlib/pyplot.py", line 2611, in imshow
    __ret = gca().imshow(
  File "/usr/local/lib/python3.9/dist-packages/matplotlib/_api/deprecation.py", line 454, in wrapper
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/matplotlib/__init__.py", line 1423, in inner
    return func(ax, *map(sanitize_sequence, args), **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/matplotlib/axes/_axes.py", line 5577, in imshow
    im._scale_norm(norm, vmin, vmax)
  File "/usr/local/lib/python3.9/dist-packages/matplotlib/cm.py", line 405, in _scale_norm
    raise ValueError(
ValueError: Passing a Normalize instance simultaneously with vmin/vmax is not supported.  Please pass vmin/vmax directly to the norm when creating it.
`

**To Reproduce**
Steps to reproduce the behavior:
1. Go to scripts/
2. Run propagation_heatmaps.sh
3. After a few seconds, script fails

**Expected behavior**
Script should generate images with propagation heat maps.

**Screenshots**
[image]

**Additional info:**
 - OS: Debian 5.10.140-1 (2022-09-02)
 - Version v2.3.3

coulisse, 2022-09-26:
Hello 
which version of python, matplotlib, numpy are you using

Thanks
Corrado

minitajfun, 2022-09-26:
python 3.9.2
matplotlib 3.6.0
numpy 1.23.3

coulisse, 2022-09-28:
I tested it with matplotlib 3.4.0.
Could you use this version in order to check if it runs?

coulisse, 2022-09-28:
I made a test and I confirm that the problem is related to matplotlib 3.6.0.
A workaround is to downgrade it to 3.4.

` pip install matplotlib==3.4 `
or
` pip3 install matplotlib==3.4 `

I'll work to solve the problem.

coulisse, 2022-09-28:
solved with new versiom 2.3.4

closed 2022-09-28 by coulisse

coulisse, 2023-01-02:
Hello, with new version 2.4, matplotlib is no longer used