coulisse/spiderweb #25: Proposal for improvements

s50u, 2022-12-18:
<html><body>
<p>After upgrading to the latest spiderweb version 2.3.4 I have some suggestions for improvements.</p>

<p>What is QSO and what is DX SPOT?</p>

<p>- QSO is a contact between two amateur radio operators.<br>
      - DX SPOT is a piece of information sent from one station to every other one logged in on the DX Cluster in real time.</p>
    
<p>Under this definition I propose a change a QSO to DX SPOTS in Graphs & stats plots page as:</p>

<p>1. /lib/qso_hour_band.py</p>

<p>plt.suptitle("DX SPOTS per hour in last month")<br>
      plt.ylabel("DX SPOTS")</p>
    
<p>2. /lib/qso_monts.py</p>
    
<p>plt.suptitle("DX SPOTS per month")<br>
      plt.ylabel("DX SPOTS")<br>
    </p>
    
<p>3. /lib/qso_trend.py</p>

<p>plt.suptitle("DX SPOTS trend")<br>
      plt.ylabel("DX SPOTS")</p>

<p>4. /lib/qso_world_map.py</p>

<p>plt.suptitle("World DX SPOTS in last month")</p>

<p>Connected nodes title may be replaced with "Physically connected callsigns to ?node_callsign?" because we show both nodes and users listed</p>

<p>5. /templates/plots.html</p>

<p>&lt;h3 class="text-center"&gt;Physically connected callsigns to
      &lt;_node_callsign_&gt;&lt;/h2&gt;</p>

<p>Seems 'Callsign' field is limited to 8 characters, anything more than that is cut out. An increase to 12 characters might be sufficient.</p>

<p>6. DIGI filter</p>

<p>DIGI filtering is not working as expected some FT4 are left to pass through. I would like to suggest exactly the same filter as for FT8 spots. Filtering by spot comment field text "FT4" and by FT4 frequencies.</p>

160m |
80m | 3.575
60m |  
40m | 7.0475
30m | 10.140
20m | 14.080
17m | 18.104
15m | 21.140
12m | 24.919
10m | 28.180
6m | 50.318

Thanks, Dan S50U

</body>
</html>

coulisse, 2022-12-18:
Thanks Dan for your suggestions, I very appreciate them! I'm working now at a new version and I'm just reviewing the graph page

s50u, 2022-12-18:
Cool. &#128077;
Thank you for all the effort you put into the spiderweb app.

Ciao, Danilo S50U

s50u, 2022-12-18:
It would be worthwhile to consider and for security reasons to exclude the ip number in the "Link" field from the table on "Physically connected callsigns to _node_call_" plots.html page.

coulisse, 2022-12-24:
A first prototype is here: [test.iu1bow.it](https://test.iu1bow.it)
I made a first graph (and is not completed...)

s50u, 2022-12-29:
propagation_heatmaps.sh

1. I would change Your Region is to Your Continent is:

2. explanatory text should be added somewhere e.g.
Based on DX SPOTS from stations in <your_continent_selection "e.g. Europe"> during the last 15 minutes displayed by Continent and Band

coulisse, 2022-12-30:
Hello, I made some changes, you can find on my [test site](https://test.iu1bow.it).
Concerning the **world dx spot** in last mont, what do you think if I change it in **dx spot in last hour (_or last day_)** in order to see which is the current activity?

I have yet to change the digi filter for ft4.

Waiting for your feedback.

coulisse, 2022-12-31:
Concerning FT4 the frequency you reported 

160m |
80m | 3.575
60m |  
40m | 7.0475
30m | 10.140
20m | 14.080
17m | 18.104
15m | 21.140
12m | 24.919
10m | 28.180
6m | 50.318

are already in the 
[modes.json](https://github.com/coulisse/spiderweb/files/10326907/modes.zip)

Could you tell me any spots with this issue (e.g IU1BOW to S50U on 31/12/2022 at 7:35)?

coulisse, 2022-12-31:
Concerning the callsign length is not limited to 8 character (see below)

[image]

Could you tell me any spots with this issue ?
Thanks for your collaboration!
Regards and **Happy new year!!!**

s50u, 2022-12-31:
_> world dx spot in last mont, what do you think if I change it in dx spot in last hour (or last day)_

Yes, maybe it would be really better to see the statistics of active stations last hour on **world dx spot** for for current activity.

_> DX SPOTS per hour in last month_

Wouldn't it be better to show cycle hourly graph around the **clock with 00:00 at the top**? 

_> Concerning **FT4** the frequency you reported  .... Could you tell me any spots with this issue_

I set filter to CW and PHONE only on your test page. The filter seems to work fine for FT4 spots and the filter doesn't work on my older spiderweb version correctly. However, the spot is leaked through the filter if the spot is not spotted in the FTx frequency range. In this case we would add filtering of DIGI spots also in the comment field (FTx). Example:

EA5GXE 7.000 OZ1ANE Denmark **FT8** - tnx qso 73

_> Concerning the callsign length is not limited to 8 character_

No, this was not meant for spots on DX Cluster page but for "Physically connected callsigns to IU1BOW" list in the **Callsign** field.

**Wishing you and yours health and prosperity in the new year**,

HNY 2023, Danilo S50U

s50u, 2022-12-31:
_> However, the spot is leaked through the filter if the spot is not spotted in the FTx frequency range. In this case we would add filtering of DIGI spots also in the comment field (FTx)._

Adding a search string also by word filtering in the Comments field (e.g. FT) should be flexible. Let's say we can add also a word MSK, PSK ... and other **DIGI** modes.

s50u, 2022-12-31:
A non-breaking, fixed space it wandered at the beginning of the **Band activity** plot explanation text, see:

**&nbsp** Based on DX SPOTS from stations in Europe during the last 15 minutes, displayed by Continent and Band

s50u, 2023-01-01:
Good morning into new year.

Would you also add a Maximum Usable Frequency MUF plot that provides near-realtime maps and data about ionospheric conditions generated every 15 minutes. See,

https://prop.kc2g.com/renders/current/mufd-normal-now.svg

For more information look here https://prop.kc2g.com/

Danilo S50U

s50u, 2023-01-01:
Or maybe even better this one https://dr2w.de/dx-propagation/

Danilo S50U

coulisse, 2023-01-01:
Hello
I have created issue #26 and issue #27 in order to separate the request.
However I released the last version on my test site and i ported it to master branch of git.

If all ok tomorrow I will make the package and I'll close  the issue #25

Best regards
Corrado

coulisse, 2023-01-02:
Closed with release 2.4

closed 2023-01-02 by coulisse