coulisse/spiderweb #26: add filtering of DIGI spots also in the comment field.

coulisse, 2022-12-31:
Adding a search string also by word filtering in the Comments field (e.g. FT) should be flexible. Let's say we can add also a word MSK, PSK ... and other **DIGI** modes.

_Originally posted by @s50u in https://github.com/coulisse/spiderweb/issues/25#issuecomment-1368182639_

s50u, 2022-12-31:
I hope we understand each other? DIGI filter should not only filter spots **by frequency range** but also filter spots **by the content** of the string _digi modes_ in the comment field of the spot. I recall some commonly used _digi modes_ tags such as:

- FT8 and FT4
- AMTOR
- D-STAR
- FELD-HELL
- MT63
- MFSK441 and OLIVIA MFSK
- JT6M and JT65
- JS8
- AX25
- AMPR
- APRS
- PACTOR
- AMTOR
- PSK31
- QPSK31 and QPSK63
- RTTY

I think that by eliminating these digi modes from the comment field of the spots we would eliminate a lot of spots that are not in the digi modes frequency range filter.

Danilo S50U

HB9VQQ, 2023-04-28:
Comment field Filtering needs to be as flexible as possible and not limited to only filtering Digi Modes but any user-entered expression/words. 

Use cases:
Only show Spots with "FT8" in the comment
Only show Spots with "FT8" and "CQ" in the comment
Only show Spots with "FT8" or "CW" in the comment

Do not show Spots with "POTA" in the comment
Do not show Spots with "CW" and "QRP" in the comment
Do not show Spots with "FT8" or "CW" in the comment

Thanks