coulisse/spiderweb #27: MUF Charts

coulisse, 2023-01-01:
Good morning into new year.

Would you also add a Maximum Usable Frequency MUF plot that provides near-realtime maps and data about ionospheric conditions generated every 15 minutes. See,

https://prop.kc2g.com/renders/current/mufd-normal-now.svg

For more information look here https://prop.kc2g.com/

Or maybe even better this one https://dr2w.de/dx-propagation/

Danilo S50U

_Originally posted by @s50u in https://github.com/coulisse/spiderweb/issues/25#issuecomment-1368395824_

s50u, 2023-01-02:
GM. Maybe plots.html could be used only for the graph SPOTS statistics and the MUF conditions page could be separated into its own html page.

Good luck, Danilo

s50u, 2023-01-20:
or if we just use the python3 package for propagations plots https://www.qsl.net/hz1jw/pythonprop/index.html

GL Danilo S50U

HB9VQQ, 2023-02-20:
If you like you can use my animated MUF map and integrate it 
https://muf.hb9vqq.ch/

HB9VQQ, 2023-02-21:
And while we are at Solar/Ionospheric KPIs here is my Grafana Dashboard which you can use if required 
https://grafana.gafner.net/goto/zUdMArJVk?orgId=1

coulisse, 2023-02-27:
Very interesting and beautifull

coulisse, 2023-11-14:
Hello I see that you implemented it: I will port on my branch this feature from your branch, that is very good!
https://webcluster.hb9vqq.ch/propagation.html

coulisse, 2023-11-18:
Using this types of external websites, there are too many issues (see report below):
[Lighthouse Report Viewer.pdf](https://github.com/coulisse/spiderweb/files/13399599/Lighthouse.Report.Viewer.pdf)

We need to find another way

Corrado