coulisse/spiderweb #32: requirements costraint

manolab, 2023-01-24:
I had to make that modification on requirements.txt, otherwise pip couldn't resolve conflicts:

root@cluster:/opt/spiderweb# git diff
diff --git a/requirements.txt b/requirements.txt
index 4cd9eee..9e82c0b 100644
--- a/requirements.txt
+++ b/requirements.txt
@@ -17,7 +17,7 @@ lazy-object-proxy==1.9.0
 lesscpy==0.15.1
 MarkupSafe==2.1.1
 mccabe==0.7.0
-mysql-connector-python==8.0.31
+mysql-connector-python
 numpy==1.24.1
 pandas==1.5.2
 platformdirs==2.6.2

Is it me?

coulisse, 2023-01-24:
I forced to used the version 8.0.31 for security constraints.
Which is the version of your mysql-connector-python ?

thanks
Corrado

manolab, 2023-01-24:
```
# pip3 show mysql-connector-python
Name: mysql-connector-python
Version: 8.0.12
Summary: MySQL driver written in Python
Home-page: http://dev.mysql.com/doc/connector-python/en/index.html
Author: Oracle and/or its affiliates
Author-email: UNKNOWN
License: GNU GPLv2 (with FOSS License Exception)
Location: /usr/local/lib/python3.9/dist-packages
Requires: protobuf
Required-by: 
```

coulisse, 2023-01-24:
The version 8.0.12 has a security issue, I fix it upgrading to 8.0.31 in august 2022.

See [https://security.snyk.io/package/pip/mysql-connector-python/8.0.12](https://security.snyk.io/package/pip/mysql-connector-python/8.0.12)

So I preferr to mantain this version.

manolab, 2023-01-25:
I see but pip couldn't resolve the constraints. I'll try to reproduce the issue and send you the output of the pip command.

manolab, 2023-01-25:
```
$ pip3 install -r requirements.txt 
Requirement already satisfied: astroid==2.12.14 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 1)) (2.12.14)
Requirement already satisfied: charset-normalizer==2.1.1 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 2)) (2.1.1)
Requirement already satisfied: click==8.1.3 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 3)) (8.1.3)
Requirement already satisfied: dill==0.3.6 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 4)) (0.3.6)
Requirement already satisfied: docopt-ng==0.8.1 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 5)) (0.8.1)
Requirement already satisfied: easywatch==0.0.5 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 6)) (0.0.5)
Requirement already satisfied: Flask==2.2.2 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 7)) (2.2.2)
Requirement already satisfied: Flask-Minify==0.41 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 8)) (0.41)
Requirement already satisfied: Flask-WTF==1.0.1 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 9)) (1.0.1)
Requirement already satisfied: htmlmin==0.1.12 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 10)) (0.1.12)
Requirement already satisfied: idna==3.4 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 11)) (3.4)
Requirement already satisfied: isort==5.11.4 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 12)) (5.11.4)
Requirement already satisfied: itsdangerous==2.1.2 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 13)) (2.1.2)
Requirement already satisfied: Jinja2==3.1.2 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 14)) (3.1.2)
Requirement already satisfied: jsmin==3.0.1 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 15)) (3.0.1)
Requirement already satisfied: lazy-object-proxy==1.9.0 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 16)) (1.9.0)
Requirement already satisfied: lesscpy==0.15.1 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 17)) (0.15.1)
Requirement already satisfied: MarkupSafe==2.1.1 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 18)) (2.1.1)
Requirement already satisfied: mccabe==0.7.0 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 19)) (0.7.0)
Collecting mysql-connector-python==8.0.31
  Using cached mysql_connector_python-8.0.31-cp39-cp39-manylinux1_x86_64.whl (23.5 MB)
Requirement already satisfied: numpy==1.24.1 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 21)) (1.24.1)
Requirement already satisfied: pandas==1.5.2 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 22)) (1.5.2)
Requirement already satisfied: platformdirs==2.6.2 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 23)) (2.6.2)
Requirement already satisfied: ply==3.11 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 24)) (3.11)
Requirement already satisfied: protobuf==4.21.12 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 25)) (4.21.12)
Requirement already satisfied: python-dateutil==2.8.2 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 26)) (2.8.2)
Requirement already satisfied: pytz==2022.7 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 27)) (2022.7)
Requirement already satisfied: rcssmin==1.1.1 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 28)) (1.1.1)
Requirement already satisfied: requests==2.28.1 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 29)) (2.28.1)
Requirement already satisfied: six==1.16.0 in /usr/lib/python3/dist-packages (from -r requirements.txt (line 30)) (1.16.0)
Requirement already satisfied: tomlkit==0.11.6 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 31)) (0.11.6)
Requirement already satisfied: urllib3==1.26.13 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 32)) (1.26.13)
Requirement already satisfied: watchdog==2.2.0 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 33)) (2.2.0)
Requirement already satisfied: Werkzeug==2.2.2 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 34)) (2.2.2)
Requirement already satisfied: wrapt==1.14.1 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 35)) (1.14.1)
Requirement already satisfied: WTForms==3.0.1 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 36)) (3.0.1)
Requirement already satisfied: xxhash==3.1.0 in /home/sysop/.local/lib/python3.9/site-packages (from -r requirements.txt (line 37)) (3.1.0)
Requirement already satisfied: typing-extensions>=3.10 in /home/sysop/.local/lib/python3.9/site-packages (from astroid==2.12.14->-r requirements.txt (line 1)) (4.4.0)
[... cut ...]
ERROR: Cannot install -r requirements.txt (line 20) and protobuf==4.21.12 because these package versions have conflicting dependencies.

The conflict is caused by:
    The user requested protobuf==4.21.12
    mysql-connector-python 8.0.31 depends on protobuf<=3.20.1 and >=3.11.0

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/user_guide/#fixing-conflicting-dependencies
```

coulisse, 2023-01-25:
Could you send me the output of your environment, with command "pip freeze" or "pip3 freeze"?

So  I could riproduce your environment.

Thanks 
Corrado

manolab, 2023-01-26:
```
$ pip3 freeze
astroid==2.12.14
attrs==21.4.0
bjoern==3.1.0
certifi==2021.10.8
chardet==4.0.0
charset-normalizer==2.1.1
click==8.1.3
click-plugins==1.1.1
cligj==0.7.2
cryptography==3.3.2
cycler==0.11.0
dill==0.3.6
docopt-ng==0.8.1
easywatch==0.0.5
Fiona==1.8.21
Flask==2.2.2
Flask-Minify==0.41
Flask-WTF==1.0.1
geopandas==0.10.2
htmlmin==0.1.12
httplib2==0.18.1
idna==3.4
importlib-metadata==6.0.0
isort==5.11.4
itsdangerous==2.1.2
Jinja2==3.1.2
jsmin==3.0.1
kiwisolver==1.3.2
lazy-object-proxy==1.9.0
lesscpy==0.15.1
MarkupSafe==2.1.1
matplotlib==3.4.3
mccabe==0.7.0
munch==2.5.0
mysql-connector-python==8.0.12
numpy==1.24.1
pandas==1.5.2
patsy==0.5.2
Pillow==9.0.1
platformdirs==2.6.2
ply==3.11
protobuf==4.21.12
pycurl==7.43.0.6
pyOpenSSL==20.0.1
pyparsing==3.0.5
pyproj==3.3.0
PySimpleSOAP==1.16.2
python-apt==2.2.1
python-dateutil==2.8.2
python-debian==0.1.39
python-debianbts==3.1.0
pytz==2022.7
PyYAML==6.0
rcssmin==1.1.1
reportbug==7.10.3+deb11u1
requests==2.28.1
scipy==1.7.2
Shapely==1.8.0
six==1.16.0
staticjinja==4.1.1
statsmodels==0.13.0
tomlkit==0.11.6
typing-extensions==4.4.0
urllib3==1.26.13
watchdog==2.2.0
Werkzeug==2.2.2
wrapt==1.14.1
WTForms==3.0.1
xxhash==3.1.0
zipp==3.11.0
```

DocViper666, 2023-08-06:
FYI on a actual Ubuntu 22.04 LTS or so Python 3.11 and pip are not compatible with the provided requirements.txt

I would recommend to use `>=` so that on newer systems the latest available version is installed by pip. A fixed restriction is not good to patch security issues.

```python
astroid>=2.12.14
charset-normalizer>=2.1.1
click>=8.1.3
dill>=0.3.6
docopt-ng>=0.8.1
easywatch>=0.0.5
Flask>=2.2.2
Flask-Consent>=0.0.3
Flask-Minify>=0.41
Flask-WTF>=1.0.1
htmlmin>=0.1.12
idna>=3.4
isort>=5.11.4
itsdangerous>=2.1.2
Jinja2>=3.1.2
jsmin>=3.0.1
lazy-object-proxy>=1.9.0
lesscpy>=0.15.1
MarkupSafe>=2.1.1
mccabe>=0.7.0
mysql-connector-python>=8.0.31
numpy>=1.24.1
pandas>=1.5.2
platformdirs>=2.6.2
ply>=3.11
protobuf>=4.21.12
python-dateutil>=2.8.2
pytz>=2022.7
rcssmin>=1.1.1
requests>=2.28.1
six>=1.16.0
tomlkit>=0.11.6
urllib3>=1.26.13
watchdog>=2.2.0
Werkzeug>=2.2.2
wrapt>=1.14.1
WTForms>=3.0.1
xxhash>=3.1.0
```

coulisse, 2023-11-11:
Check  with 4.2.5 release

closed 2023-11-11 by coulisse