coulisse/spiderweb #36: no data found

HB9VQQ, 2023-02-23:
Until this morning all was working found when I noticed there are no new Spots incoming. In the log I found.

2023-02-23 09:40:08,720 [WARNING] (webapp.py) no data found

Any idea?

HB9VQQ, 2023-02-23:
Apparently it is mariadb related

# journalctl -u mariadb -f
 mariadbd[109316]: 2023-02-23 10:06:21 22 [Warning] Aborted connection 22 to db: 'dxcluster' user: 'root' host: 'localhost' (Got an error reading communication packets)

coulisse, 2023-02-23:
Hello 
yes it is apparently related to mariadb.

could you check if mariadb is running (systemctl status mariadb)

Moreover I see in your message that you are connecting as root user to dxcluster.

I think is better to define a specific user and assign it to dxcluster:

for example
grant all privileges on dxcluster.* to webdb@localhost identified by 'pswd';

By the way you have to modify config.json with new user and password.

Waiting for your feedback
Best regards
Corrado

HB9VQQ, 2023-02-23:
Thanks, 

yes mariadb is running
[image]

It was running without problems using root. I will change it as soon as it is working again.

I deleted the database 'dxcluster' and was hoping it is re-created by restarting dxspider, That's not the case unfortunately. Then I tried to follow https://github.com/coulisse/spiderweb/blob/master/docs/DB_ISSUES.md but the command to create the table/coloums returned errors.

HB9VQQ, 2023-02-23:
While it is working again I still see

 [WARNING] (webapp.py) no data found

in webapp.log

closed 2023-02-23 by HB9VQQ