coulisse/spiderweb #37: [WARNING] (webapp.py) no data found

HB9VQQ, 2023-02-23:
I see this warning in webapp.log

[WARNING] (webapp.py) no data found

Disregard?

coulisse, 2023-02-23:
Hello
It seems that it could not connect to database.

Could you connect manually to databse dxcluster and see if table spot is empty?

HB9VQQ, 2023-02-23:
Database connection works.

[image]

HB9VQQ, 2023-02-23:
Just to clarify everything seems to work except seeing this warning in the log
https://webcluster.hb9vqq.ch/

coulisse, 2023-02-23:
Please, change the log level from INFO  to DEBUG in webapp_config_log.ini  and restart spiderweb, in order to have more informations on the log

HB9VQQ, 2023-02-23:
Done
[image]

coulisse, 2023-02-27:
I dont' find  errors... I think it is a "false positive", however I'll check better on new release

HB9VQQ, 2023-02-27:
Did not occure anymore. Closing issue . Thank you

closed 2023-02-27 by HB9VQQ