coulisse/spiderweb #40: FT8 Spots shown when Mode CW Filter active

HB9VQQ, 2023-04-28:
If the CW Mode Filter is active there are still FT8 Spots displayed mainly on the 6m Band 

[image]

50.313 is the FT8 QRG on the 6m Band, which needs to be excluded when the CW filter is enabled

DocViper666, 2023-08-06:
You can assign modes by changing the 
`/cfg/modes.json`
FT8 is seen as "within the CW" range. You would need to have a range for CW lower and higher seperatly - and assign the FT8 to digi. But I have not tried that myself.

coulisse, 2023-11-10:
I'll change `modes.json` in order to check all ft8 frequencies  are in DIGI modes

coulisse, 2023-11-11:
Closed with 4.2.5 release

closed 2023-11-11 by coulisse