coulisse/spiderweb #49: Error On raspi install

9M2PJU, 2024-01-11:
[image]

9M2PJU, 2024-01-15:
[image]

DocViper666, 2024-01-15:
Hi Piju,

I think there is an error in your config.json
The error states that a comma is missing in line 11 (after the string with your cluster callsign)
Please compare the structure of your file with the template:
https://github.com/coulisse/spiderweb/blob/master/cfg/config.json.template

73 Joe, OE5JFE (from the SOTA Reflector)

9M2PJU, 2024-01-23:
i have found new problem. nothing from dx spider writes to mariadb. tried many config as suggested from dx spider mailing list but nothings work.

# the SQL database DBI dsn

$dsn = "dbi:mysql:dbname=dxcluster";

$dbuser = "sysop";

$dbpass = "password";

# the SQL database DBI dsn

$dsn = "dbi:mysql:dxcluster:localhost:3306";

$dbuser = "sysop";

$dbpass = "password";

DocViper666, 2024-01-23:
Did you create the database and tables on your mysql server?
The automatic creation of the table did not work for me.

So here the sql script to create it:
https://github.com/DocViper666/dxspider-deployment/blob/master/cluster-db/init/create_new_db.sql

9M2PJU, 2024-01-23:
i have already create tables with this https://github.com/DocViper666/dxspider-deployment/blob/master/cluster-db/init/create_new_db.sql

and create user and grand all privilleges.
and set DXVars.pm to

# the SQL database DBI dsn
$dsn = "dbi:mysql:dxcluster:localhost:3306";
$dbuser = "sysop";
$dbpass = "password";

but nothing writes to the database

wifi75, 2024-02-05:
> Did you create the database and tables on your mysql server? The automatic creation of the table did not work for me.
> 
> So here the sql script to create it: https://github.com/DocViper666/dxspider-deployment/blob/master/cluster-db/init/create_new_db.sql

hello I trieded to import file create_new_db.sql with phpmyadmin but there is this problem..

[image]

how I can fix?

DocViper666, 2024-02-05:
I only see warnings and no error. So I guess it should be fine.
And it would be good to changes to English user interface for the screenshot - at least for me to understand what the error says.
 
Did you check if now the database dxcluster exists?
And it should have a table named "spot".
You can display that in phpadmin

wifi75, 2024-02-05:
> I only see warnings and no error. So I guess it should be fine. And it would be good to changes to English user interface for the screenshot - at least for me to understand what the error says.
>
> Did [...]

there is terminal after command: python3 webapp.py

[image]

from the web typing ip:5000 there is an error
[image]

below golder location
[image]

DocViper666, 2024-02-05:
The error message states that there is no data in the database. So start your DXspider on a terminal. Log into your dxcluster by telnet and create a test spot. This way you should have at least one data entry in your database.
To check if the connection works have a look in the debug log file within /spider/

wifi75, 2024-02-05:
and this error:

[image]

DocViper666, 2024-02-05:
Did you follow the readme?
https://github.com/coulisse/spiderweb#configuration
You will need to create the config file based on the template.

wifi75, 2024-02-05:
> Did you follow the readme? https://github.com/coulisse/spiderweb#configuration You will need to create the config file based on the template.

sure:

[image]

[image]

DocViper666, 2024-02-05:
See my updated comment. No issues with the error. It looks in two folders and found it in the second one.

wifi75, 2024-02-05:
a question, because in the present guide it is not clear, but where should the spiderweb folder be copied?

DocViper666, 2024-02-05:
No specific place in general. As long as it can reach the database it does not matter.

I use a docker container with bjorn to deploy it

coulisse, 2024-03-10:
I close this issue since is not related to spiderweb but to Dxspider.
I'm thinking to switch to Sqllite in next versions

closed 2024-03-10 by coulisse