coulisse/spiderweb #51: Minor v2.5.2 corrections

s50u, 2024-01-12:
1. _/templates/plots.html_

Change line 52 from: strong>Physically connected callsigns to {{ mycallsign }}</strong
to: <strong>**{{ mycallsign }} telnet nodes & users online: Nodes counts.get('NODE') Users counts.get('USER')**</strong>
reading data from "who" table in "Type" column.

2. _/templates/plots.html_

Increase the number of characters in the table in the "Callsign" field to **12 characters**. currently eight characters
are allocated and anything more than that is cut out. 

3. _Footer_

Add simple **visit counter service** to count visitor as unique visit only once per 12 or 24 hours.
© Copyleft: 2020-2024 IU1BOW - Spiderweb v2.5.2 | website hits {{visit_counter}} since 1st January 2024

5. _Browser cache_

Is **web browser caching** to be improved for plots and propagation html pages or is it just me at **[S50CLX](https://s50clx.infrax.si)**. Ctrl-f5 works but it is somehow annoying. FF 121.0.1

Ciao, Dan S50U

coulisse, 2024-03-10:
The  web browser was improved with v2.5.3

coulisse, 2024-03-23:
For the point n.2 I couldn't find the problem. Could you give me a screen ?

In attachment my screenshot
[image]

coulisse, 2024-03-23:
For visitor counter there is a specific issue #56

s50u, 2024-03-23:
I was connected with 14 characters s50uuuuuuuu-18 which is probably not a very common case. I saw Callsigns with twelve characters most.

[image]

coulisse, 2024-03-30:
Could you test my [pre-release v2.5.4](https://github.com/coulisse/spiderweb/archive/refs/tags/v2.5.4.zip)  ?

coulisse, 2024-03-31:
Released version v2.5.4

closed 2024-03-31 by coulisse