coulisse/spiderweb #53: Error on installation

maserowik, 2024-01-29:
I followed the guide and when when i issue the python3 webapp.py i receive the following message

sysop@DX-Spider:~/spider/spiderweb $ python3 webapp.py
Traceback (most recent call last):
  File "/home/sysop/spider/spiderweb/webapp.py", line 13, in <module>
    from lib.qry import query_manager
  File "/home/sysop/spider/spiderweb/lib/qry.py", line 10, in <module>
    import pandas as pd
  File "/home/sysop/.local/lib/python3.9/site-packages/pandas/__init__.py", line 16, in <module>
    raise ImportError(
ImportError: Unable to import required dependencies:
numpy: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.
sysop@DX-Spider:~/spider/spiderweb $

Thanks
Mike N3BSQ

coulisse, 2024-02-01:
The problem is with numpy standard library..

could you try to install it manually with
`pip install numpy`

See also this: [https://github.com/numpy/numpy/issues/15669](https://github.com/numpy/numpy/issues/15669)

Waiting for your feedback
best regards
Corrado

DocViper666, 2024-02-05:
2024-02-01 14:20:14,676 [WARNING] (plot_data_provider.py) no data found

No data is found so it crashes.

Add a spot to your cluster (login via telnet)
Place spot:
`DX callsign 7100 test`

Check if it wrote to the db. Then try again

9M2PJU, 2024-03-11:
i have similar problem,
for numpy, installing libopenblas-dev solved the problem.
but when running python3 webapp.py, same error as him.

we need to clarify about DXVars.pm, 
which setting allow make DX spider writes to database?
currently my setting is 

[image]

i think i didnt write to mariadb

DocViper666, 2024-03-11:
Pls understand what you define here:

```
CLUSTER_DSN="dbi:mysql:dxcluster:cluster-db:3306"

dbi = database interface
mysql = language of the database
dxcluster = name of the database
cluster-db = hostname of database
3306 = port at hostname
```

So in your example I don't think the instructions suggest sysop as name of the database?

9M2PJU, 2024-03-11:
ok i have changed user and db name to spiderweb, same error happened. it still crashed.

DocViper666, 2024-03-11:
I will give up soon...

So your database name is spiderweb:

CLUSTER_DSN="dbi:mysql:spiderweb:localhost:3306"

and you have set your username and password correctly for the mysql database?

Did you try to post a spot on your cluster? Is it running or are you at a test setup?

You have no data so how should the spiderweb show anything?

Loging to your dxspider cluster and type:

`DX 9M2PJU 7100 test`

9M2PJU, 2024-03-12:
> I will give up soon...
> 
> So your database name is spiderweb:
> 
> CLUSTER_DSN="dbi:mysql:spiderweb:localhost:3306"
> 
> and you have set your username and password correctly for the mysql database? All the settings need to match. In dxspider, in the mariadb for the database spiderweb and also in the settings of spiderweb.
> 
> Did you try to post a spot on your cluster? Is it running or are you at a test setup?
> 
> You have no data so how should the spiderweb show anything?
> 
> Loging to your dxspider cluster and type:
> 
> `DX 9M2PJU 7100 test`

yes. username and passwd are correct and i also tried to post a spot. i read earlier replies on this thread. but still, nothing happens and there is nothing on server_IP:5000

coulisse, 2024-03-17:
check the .dat files inside /home/sysop/spider/local_data/debug/2024

Inside these file you can find connection error to DBI.

coulisse, 2024-03-30:
Closing this issue due to lack of recent activity. Feel free to reopen if needed.

closed 2024-03-30 by coulisse