coulisse/spiderweb #55: mysql version not compatible?

wifi75, 2024-02-07:
hello I have installed spiderweb on ubuntu 22.04 with mysql 8.0.36, see below:
[image]

after start test this is error from webapp.log

```
2024-02-07 07:07:16,578 [INFO] (webapp.py) Starting SPIDERWEB
2024-02-07 07:07:16,579 [INFO] (webapp.py) Version:v2.5.2
2024-02-07 07:07:16,580 [INFO] (cty.py) CTY: start initialization
2024-02-07 07:07:16,580 [INFO] (cty.py) /home/sysop/spiderweb/lib/../static/data/cty_wt_mod.dat updated (1.0 days), is not necessary to download it
2024-02-07 07:07:16,583 [INFO] (cty.py) number of lines reads: 8536
2024-02-07 07:07:16,583 [INFO] (cty.py) number of valid lines: 7561
2024-02-07 07:07:16,660 [INFO] (cty.py) number of entities: 346
2024-02-07 07:07:16,661 [INFO] (cty.py) number of single alias: 28410
2024-02-07 07:07:17,222 [INFO] (cty.py) memory used for prefix: 1310792 bytes
2024-02-07 07:07:17,222 [INFO] (cty.py) CTY: initialization complete
2024-02-07 07:07:17,222 [INFO] (qry.py) [Errno 2] No such file or directory: '../cfg/config.json'
2024-02-07 07:07:17,222 [INFO] (qry.py) trying with other path...
2024-02-07 07:07:17,223 [INFO] (qry.py) config file loaded
2024-02-07 07:07:17,237 [INFO] (qry.py) db connection pool created
2024-02-07 07:07:17,237 [INFO] (adxo.py) connection to: http://dxcal.kj4z.com/dxcal
2024-02-07 07:07:17,551 [INFO] (adxo.py) number ADXO events: 0
2024-02-07 07:07:17,552 [INFO] (plot_data_provider.py) Class: ContinentsBandsProvider init start
2024-02-07 07:07:17,552 [INFO] (plot_data_provider.py) Class: ContinentsBandsProvider refresh data
2024-02-07 07:07:17,552 [INFO] (plot_data_provider.py) Start
2024-02-07 07:07:17,552 [INFO] (plot_data_provider.py) doing query...
2024-02-07 07:07:17,555 [INFO] (plot_data_provider.py) query done
2024-02-07 07:07:17,556 [INFO] (plot_data_provider.py) Class: ContinentsBandsProvider init end
2024-02-07 07:07:17,556 [INFO] (plot_data_provider.py) Class: SpotsPerMounthProvider init start
2024-02-07 07:07:17,556 [INFO] (plot_data_provider.py) Class: SpotsPerMounthProvider refresh data
2024-02-07 07:07:17,556 [INFO] (plot_data_provider.py) Start
2024-02-07 07:07:17,556 [INFO] (plot_data_provider.py) doing query...
2024-02-07 07:07:17,556 [ERROR] (qry.py) 1064 (42000): You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'int) as current_year,
			cast(sum(
				case 
					when YEAR(s1.ym)=YEAR(now())-1' at line 8
2024-02-07 07:07:17,557 [INFO] (plot_data_provider.py) query done
2024-02-07 07:07:17,557 [INFO] (plot_data_provider.py) Class: SpotsPerMounthProvider init end
2024-02-07 07:07:17,557 [INFO] (plot_data_provider.py) Class: SpotsTrend init start
2024-02-07 07:07:17,558 [INFO] (plot_data_provider.py) Class: SpotsTrend refresh data
2024-02-07 07:07:17,558 [INFO] (plot_data_provider.py) Start
2024-02-07 07:07:17,558 [INFO] (plot_data_provider.py) doing query...
2024-02-07 07:07:17,563 [INFO] (plot_data_provider.py) query done
2024-02-07 07:07:17,567 [INFO] (plot_data_provider.py) Class: SpotsTrend init end
2024-02-07 07:07:17,567 [INFO] (plot_data_provider.py) Class: HourBand init start
2024-02-07 07:07:17,567 [INFO] (plot_data_provider.py) Class: HourBand refresh data
2024-02-07 07:07:17,567 [INFO] (plot_data_provider.py) Start
2024-02-07 07:07:17,567 [INFO] (plot_data_provider.py) doing query...
2024-02-07 07:07:17,568 [ERROR] (qry.py) 1055 (42000): Expression #3 of SELECT list is not in GROUP BY clause and contains nonaggregated column 'dxcluster.spot.freq' which is not functionally dependent on columns in GROUP BY clause; this is incompatible with sql_mode=only_full_group_by
2024-02-07 07:07:17,569 [INFO] (plot_data_provider.py) query done
```

coulisse, 2024-02-08:
Could you check this thread?

[https://stackoverflow.com/questions/41887460/select-list-is-not-in-group-by-clause-and-contains-nonaggregated-column-inc](https://stackoverflow.com/questions/41887460/select-list-is-not-in-group-by-clause-and-contains-nonaggregated-column-inc)

Tell me if it solve the problem

Thanks
Corrado

wifi75, 2024-02-08:
Hi Corrado I solved it by installing MariaDB

closed 2024-02-11 by coulisse