coulisse/spiderweb #59: not workinf after update to 2.5.4

wifi75, 2024-04-01:
hello this is error after update to last version 2.5.4
why?

```
 sudo systemctl restart spiderweb.service
root@dxspider:/home/sysop# sudo systemctl status spiderweb.service
× spiderweb.service - bjoern instance spiderweb
     Loaded: loaded (/etc/systemd/system/spiderweb.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Mon 2024-04-01 20:51:45 UTC; 5s ago
    Process: 1136 ExecStart=/usr/bin/python3 /home/sysop/spiderweb/wsgi.py (code=exited, status=1/FAILURE)
   Main PID: 1136 (code=exited, status=1/FAILURE)
        CPU: 228ms

Apr 01 20:51:45 dxspider python3[1136]: Traceback (most recent call last):
Apr 01 20:51:45 dxspider python3[1136]:   File "/home/sysop/spiderweb/wsgi.py", line 2, in <module>
Apr 01 20:51:45 dxspider python3[1136]:     from webapp import app
Apr 01 20:51:45 dxspider python3[1136]:   File "/home/sysop/spiderweb/webapp.py", line 12, in <module>
Apr 01 20:51:45 dxspider python3[1136]:     from lib.adxo import get_adxo_events
Apr 01 20:51:45 dxspider python3[1136]:   File "/home/sysop/spiderweb/lib/adxo.py", line 10, in <module>
Apr 01 20:51:45 dxspider python3[1136]:     import feedparser
Apr 01 20:51:45 dxspider python3[1136]: ModuleNotFoundError: No module named 'feedparser'
Apr 01 20:51:45 dxspider systemd[1]: spiderweb.service: Main process exited, code=exited, status=1/FAILURE
Apr 01 20:51:45 dxspider systemd[1]: spiderweb.service: Failed with result 'exit-code'.
```

coulisse, 2024-04-02:
You need to install feedparser:

pip install feedparser

wifi75, 2024-04-03:
ok now this message:

```
sudo systemctl status spiderweb.service
× spiderweb.service - bjoern instance spiderweb
     Loaded: loaded (/etc/systemd/system/spiderweb.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2024-04-03 06:36:10 UTC; 1s ago
    Process: 8543 ExecStart=/usr/bin/python3 /home/sysop/spiderweb/wsgi.py (code=exited, status=1/FAILURE)
   Main PID: 8543 (code=exited, status=1/FAILURE)
        CPU: 208ms

Apr 03 06:36:10 dxspider python3[8543]: Traceback (most recent call last):
Apr 03 06:36:10 dxspider python3[8543]:   File "/home/sysop/spiderweb/wsgi.py", line 2, in <module>
Apr 03 06:36:10 dxspider python3[8543]:     from webapp import app
Apr 03 06:36:10 dxspider python3[8543]:   File "/home/sysop/spiderweb/webapp.py", line 13, in <module>
Apr 03 06:36:10 dxspider python3[8543]:     from lib.qry import query_manager
Apr 03 06:36:10 dxspider python3[8543]:   File "/home/sysop/spiderweb/lib/qry.py", line 6, in <module>
Apr 03 06:36:10 dxspider python3[8543]:     import mariadb as my
Apr 03 06:36:10 dxspider python3[8543]: ModuleNotFoundError: No module named 'mariadb'
Apr 03 06:36:10 dxspider systemd[1]: spiderweb.service: Main process exited, code=exited, status=1/FAILURE
Apr 03 06:36:10 dxspider systemd[1]: spiderweb.service: Failed with result 'exit-code'.
```

wifi75, 2024-04-03:
solved on debian 11
with
sudo apt-get install -y libmariadb-dev
then 
pip install -r requirements.txt

maybe you should add it in your guide or in the file
requirements.txt

closed 2024-04-05 by coulisse