coulisse/spiderweb #62: Error running

9M2PJU, 2024-04-26:
(myenv) piju@kelubi:~/spiderweb-2.5.4$ python3 webapp.py
2024-04-26 23:41:16,610 [INFO] (webapp.py) Starting SPIDERWEB
2024-04-26 23:41:16,612 [INFO] (webapp.py) Version:v2.5.4
2024-04-26 23:41:16,612 [INFO] (webapp.py) visit saved on: data/visits.json
2024-04-26 23:41:16,613 [INFO] (cty.py) CTY: start initialization
2024-04-26 23:41:16,613 [INFO] (cty.py) /home/piju/spiderweb-2.5.4/lib/../data/cty_wt_mod.dat updated (0.0 days), is not necessary to download it
2024-04-26 23:41:16,616 [INFO] (cty.py) number of lines reads: 8613
2024-04-26 23:41:16,617 [INFO] (cty.py) number of valid lines: 7641
2024-04-26 23:41:16,700 [INFO] (cty.py) number of entities: 346
2024-04-26 23:41:16,701 [INFO] (cty.py) number of single alias: 28663
2024-04-26 23:41:16,701 [INFO] (cty.py) loading:/home/piju/spiderweb-2.5.4/lib/../cfg/country.json
2024-04-26 23:41:17,158 [INFO] (cty.py) memory used for prefix: 961264 bytes
2024-04-26 23:41:17,158 [INFO] (cty.py) CTY: initialization complete
2024-04-26 23:41:17,159 [INFO] (qry.py) [Errno 2] No such file or directory: '../cfg/config.json'
2024-04-26 23:41:17,159 [INFO] (qry.py) trying with other path...
2024-04-26 23:41:17,159 [INFO] (qry.py) config file loaded
2024-04-26 23:41:17,161 [INFO] (qry.py) db connection pool created
2024-04-26 23:41:17,854 [INFO] (adxo.py) number ADXO events: 16
2024-04-26 23:41:17,856 [INFO] (plot_data_provider.py) Class: ContinentsBandsProvider init start
2024-04-26 23:41:17,856 [INFO] (plot_data_provider.py) Class: ContinentsBandsProvider refresh data
2024-04-26 23:41:17,856 [INFO] (plot_data_provider.py) Start
2024-04-26 23:41:17,856 [INFO] (plot_data_provider.py) doing query...
2024-04-26 23:41:17,857 [WARNING] (plot_data_provider.py) no data found
2024-04-26 23:41:17,857 [INFO] (plot_data_provider.py) query done
2024-04-26 23:41:17,858 [INFO] (plot_data_provider.py) Class: ContinentsBandsProvider init end
2024-04-26 23:41:17,858 [INFO] (plot_data_provider.py) Class: SpotsPerMounthProvider init start
2024-04-26 23:41:17,858 [INFO] (plot_data_provider.py) Class: SpotsPerMounthProvider refresh data
2024-04-26 23:41:17,858 [INFO] (plot_data_provider.py) Start
2024-04-26 23:41:17,858 [INFO] (plot_data_provider.py) doing query...
2024-04-26 23:41:17,859 [INFO] (plot_data_provider.py) query done
2024-04-26 23:41:17,859 [INFO] (plot_data_provider.py) Class: SpotsPerMounthProvider init end
2024-04-26 23:41:17,859 [INFO] (plot_data_provider.py) Class: SpotsTrend init start
2024-04-26 23:41:17,859 [INFO] (plot_data_provider.py) Class: SpotsTrend refresh data
2024-04-26 23:41:17,859 [INFO] (plot_data_provider.py) Start
2024-04-26 23:41:17,860 [INFO] (plot_data_provider.py) doing query...
/home/piju/spiderweb-2.5.4/lib/qry.py:77: UserWarning: pandas only supports SQLAlchemy connectable (engine/connection) or database string URI or sqlite3 DBAPI2 connection. Other DBAPI2 objects are not tested. Please consider using SQLAlchemy.
  self.__data = pd.read_sql(qs, con=cnx)
2024-04-26 23:41:17,863 [INFO] (plot_data_provider.py) query done
2024-04-26 23:41:17,863 [WARNING] (plot_data_provider.py) no data found
Traceback (most recent call last):
  File "/home/piju/spiderweb-2.5.4/webapp.py", line 162, in <module>
    line_graph_st = SpotsTrend(logger, qm)
                    ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/piju/spiderweb-2.5.4/lib/plot_data_provider.py", line 297, in __init__
    super().__init__(logger, qm, [], [])
  File "/home/piju/spiderweb-2.5.4/lib/plot_data_provider.py", line 41, in __init__
    self.refresh()
  File "/home/piju/spiderweb-2.5.4/lib/plot_data_provider.py", line 337, in refresh
    qry_data = self.__load_data()
               ^^^^^^^^^^^^^^^^^^
  File "/home/piju/spiderweb-2.5.4/lib/plot_data_provider.py", line 330, in __load_data
    df = df.rolling("30D").mean()
         ^^^^^^^^^^^^^^^^^
  File "/home/piju/spiderweb-2.5.4/myenv/lib/python3.11/site-packages/pandas/core/generic.py", line 12577, in rolling
    return Rolling(
           ^^^^^^^^
  File "/home/piju/spiderweb-2.5.4/myenv/lib/python3.11/site-packages/pandas/core/window/rolling.py", line 170, in __init__
    self._validate()
  File "/home/piju/spiderweb-2.5.4/myenv/lib/python3.11/site-packages/pandas/core/window/rolling.py", line 1877, in _validate
    self._validate_datetimelike_monotonic()
  File "/home/piju/spiderweb-2.5.4/myenv/lib/python3.11/site-packages/pandas/core/window/rolling.py", line 1921, in _validate_datetimelike_monotonic
    if self._on.hasnans:
       ^^^^^^^^^^^^^^^^
  File "properties.pyx", line 36, in pandas._libs.properties.CachedProperty.__get__
  File "/home/piju/spiderweb-2.5.4/myenv/lib/python3.11/site-packages/pandas/core/indexes/base.py", line 2840, in hasnans
    return bool(self._isnan.any())
                ^^^^^^^^^^^
  File "properties.pyx", line 36, in pandas._libs.properties.CachedProperty.__get__
  File "/home/piju/spiderweb-2.5.4/myenv/lib/python3.11/site-packages/pandas/core/indexes/base.py", line 2810, in _isnan
    return isna(self)
           ^^^^^^^^^^
  File "/home/piju/spiderweb-2.5.4/myenv/lib/python3.11/site-packages/pandas/core/dtypes/missing.py", line 178, in isna
    return _isna(obj)
           ^^^^^^^^^^
  File "/home/piju/spiderweb-2.5.4/myenv/lib/python3.11/site-packages/pandas/core/dtypes/missing.py", line 203, in _isna
    raise NotImplementedError("isna is not defined for MultiIndex")
NotImplementedError: isna is not defined for MultiIndex

9M2PJU, 2024-04-26:
config.json

{
        "mysql":{
            "host":"localhost",
            "user":"piju",
            "passwd":"piju123",
            "db":"dxcluster"
        },
        "timer":{
                        "interval":30000
        },
        "mycallsign":"9M2PJU",
    "mail":"9m2pju@hamradio.my",
    "enable_cq_filter":"n",
    "telnet": {
        "host": "localhost",
        "port": "7300",
        "user": "9M2PJU",
        "password": ""
    },
        "menu": {
          "menu_list": [
            {"label":"Charts & stats", "link": "/plots.html", "external": false},
            {"label": "Propagation", "link": "/propagation.html", "external": false},
            {"label":"Sources", "link": "https://github.com/coulisse/spiderweb/" , "external": true},
            {"label":"Cookies", "link": "/cookies.html", "external": false},
            {"label":"Privacy", "link": "/privacy.html", "external": false}
            ]
        }
}

9M2PJU, 2024-04-26:
DXVars.pm database

# the SQL database DBI dsn
$dsn = "dbi:mysql:dxcluster:localhost:3306";
$dbuser = "piju";
$dbpass = "piju123";

1;

videobuff, 2024-10-05:
I found this issue and  i have the same problem. With one manual added record, the webpage shows up but there are no records added from the mysql database because no records are added to the database. Do i have to set it in dxspider ?
I added the mysql data at the end but that's all.
```
# the SQL database DBI dsn
$dsn = "dbi:mysql:dxcluster:localhost:3306";
$dbuser = "xxxx";
$dbpass = "yyyyyyyyyy"; 
```
With these details i can logon via mysql and phpmyadmin to the database

coulisse, 2024-12-21:
> This happens if there is not one single entry in your database. @coulisse Could you please add a try catch for this case (0 entries in database) to allow the application to load and not crash on start. Thanks

Solved with release  [v2.5.7](https://github.com/coulisse/spiderweb/releases/tag/v2.5.7)

closed 2024-12-21 by coulisse