coulisse/spiderweb #65: not nodes and users show

wifi75, 2024-05-14:
Hello why does it always show me 0 nodes and 0 connected users?
I am connected to 3 nodes and have 1 user connected.

[image]

DocViper666, 2024-05-19:
Because you proable did not configure your config.json correctly. In the telnet section you must provide a valid set of parameters. These will be used to connect to your cluster by spiderweb and run the "who" command to get the list of connected users and hosts.

wifi75, 2024-05-19:
can you show me what your configuration file looks like?

> Because you proable did not configure your config.json correctly. In the telnet section you must provide a valid set of parameters. These will be used to connect to your cluster by spiderweb and run the "who" command to get the list of connected users and hosts.

wifi75, 2024-05-21:
> Because you proable did not configure your config.json correctly. In the telnet section you must provide a valid set of parameters. These will be used to connect to your cluster by spiderweb and run the "who" command to get the list of connected users and hosts.

and what do I have to do?
rin the webcluster I receive everything....

DocViper666, 2024-05-21:
It is quite simple:
Put in your connection details to your dxspider telnet:

`"telnet": {
        "host": "mysite",
        "port": "7300",
        "user": "myuser",
        "password": ""
    },`

If it is the same machine and public use
```
"telnet": {
        "host": "localhost",
        "port": "7300",
        "user": "yourclustercallsign-2",
        "password": ""
    },
```
Otherwise use a valid user and password to allow spiderweb to connect to your cluster. This will allow the system to get the list of connected users and nodes

wifi75, 2024-05-23:
> It is quite simple: Put in your connection details to your dxspider telnet:
> 
> ```
> "telnet": {
>         "host": "mysite",
>         "port": "7300",
>         "user": "myuser",
>         "password": ""
>     },
> ```
> 
> If it is the same machine and public access (no user registration mandatory) use
> 
> ```
> "telnet": {
>         "host": "localhost",
>         "port": "7300",
>         "user": "yourclustercallsign-2",
>         "password": ""
>     },
> ```
> 
> Otherwise use a valid user and password to allow spiderweb to connect to your dxspider cluster. This will allow the system to get the list of connected users and nodes to populate the list.

Maybe I didn't explain myself well, my dxspider works well and is connected to 3 nodes, I receive spots!
I also receive commercials on the webcluster,
the problem is that in the webcluster status the connected nodes and also the users are not listed. that's the only problem.

DocViper666, 2024-05-23:
And I have explained how to adjust the config.json with telnet setting that will allow Spiderweb to connect to your dxspider and an run the 'who' command to get the list of users and nodes.I don't know your dxspider so I can not give you your connection details.

wifi75, 2024-05-23:
Ok thank you, work now I had set a password that doesn't exist, on my system you need a password to send commercials but not to log in.