coulisse/spiderweb #71: Installation error

maserowik, 2024-09-18:
trying to install this na di am getting the message when i issue the install command.  Any ideas?

sysop@DX-Spider:~/spider/spiderweb $ pip install -r requirements.txt
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

maserowik, 2024-09-22:
python3 -m venv --system-site-packages venv
source venv/bin/activate

python3 -m pip install --upgrade pip
python3 -m pip install -r requirements.txt

so as long as I stay in the venv environment, the webapp starts to run but not I get this 

(venv) sysop@dx-spider-test:~/web$ python3 webapp.py
2024-09-22 12:44:28,463 [INFO] (webapp.py) Starting SPIDERWEB
2024-09-22 12:44:28,467 [INFO] (webapp.py) Version:v2.5.4
2024-09-22 12:44:28,472 [INFO] (webapp.py) visit saved on: data/visits.json
2024-09-22 12:44:28,475 [INFO] (cty.py) CTY: start initialization
2024-09-22 12:44:28,476 [INFO] (cty.py) /home/sysop/web/lib/../data/cty_wt_mod.dat not present: proceding to download it
2024-09-22 12:44:28,476 [INFO] (cty.py) connection to: https://www.country-files.com/cty/cty_wt_mod.dat
2024-09-22 12:44:29,656 [INFO] (cty.py) cty file saved in: /home/sysop/web/lib/../data/cty_wt_mod.dat
2024-09-22 12:44:29,664 [INFO] (cty.py) number of lines reads: 8609
2024-09-22 12:44:29,666 [INFO] (cty.py) number of valid lines: 7638
2024-09-22 12:44:29,948 [INFO] (cty.py) number of entities: 346
2024-09-22 12:44:29,948 [INFO] (cty.py) number of single alias: 28652
2024-09-22 12:44:29,949 [INFO] (cty.py) loading:/home/sysop/web/lib/../cfg/country.json
2024-09-22 12:44:30,782 [INFO] (cty.py) memory used for prefix: 961264 bytes
2024-09-22 12:44:30,783 [INFO] (cty.py) CTY: initialization complete
2024-09-22 12:44:30,785 [INFO] (qry.py) [Errno 2] No such file or directory: '../cfg/config.json'
2024-09-22 12:44:30,786 [INFO] (qry.py) trying with other path...
2024-09-22 12:44:30,786 [INFO] (qry.py) config file loaded
2024-09-22 12:44:30,910 [INFO] (qry.py) db connection pool created
2024-09-22 12:44:31,180 [ERROR] (adxo.py) time data 'Dec 26, 2024 2024-Jan 4, 2025' does not match format '%b %d %Y'
2024-09-22 12:44:31,184 [INFO] (plot_data_provider.py) Class: ContinentsBandsProvider init start
2024-09-22 12:44:31,185 [INFO] (plot_data_provider.py) Class: ContinentsBandsProvider refresh data
2024-09-22 12:44:31,185 [INFO] (plot_data_provider.py) Start
2024-09-22 12:44:31,186 [INFO] (plot_data_provider.py) doing query...
2024-09-22 12:44:31,257 [WARNING] (plot_data_provider.py) no data found
2024-09-22 12:44:31,257 [INFO] (plot_data_provider.py) query done
2024-09-22 12:44:31,261 [INFO] (plot_data_provider.py) Class: ContinentsBandsProvider init end
2024-09-22 12:44:31,262 [INFO] (plot_data_provider.py) Class: SpotsPerMounthProvider init start
2024-09-22 12:44:31,263 [INFO] (plot_data_provider.py) Class: SpotsPerMounthProvider refresh data
2024-09-22 12:44:31,263 [INFO] (plot_data_provider.py) Start
2024-09-22 12:44:31,263 [INFO] (plot_data_provider.py) doing query...
2024-09-22 12:44:31,281 [INFO] (plot_data_provider.py) query done
2024-09-22 12:44:31,282 [INFO] (plot_data_provider.py) Class: SpotsPerMounthProvider init end
2024-09-22 12:44:31,283 [INFO] (plot_data_provider.py) Class: SpotsTrend init start
2024-09-22 12:44:31,283 [INFO] (plot_data_provider.py) Class: SpotsTrend refresh data
2024-09-22 12:44:31,284 [INFO] (plot_data_provider.py) Start
2024-09-22 12:44:31,284 [INFO] (plot_data_provider.py) doing query...
/home/sysop/web/lib/qry.py:77: UserWarning: pandas only supports SQLAlchemy connectable (engine/connection) or database string URI or sqlite3 DBAPI2 connection. Other DBAPI2 objects are not tested. Please consider using SQLAlchemy.
  self.__data = pd.read_sql(qs, con=cnx)
2024-09-22 12:44:31,409 [INFO] (plot_data_provider.py) query done
2024-09-22 12:44:31,410 [WARNING] (plot_data_provider.py) no data found
Traceback (most recent call last):
  File "/home/sysop/web/webapp.py", line 162, in <module>
    line_graph_st = SpotsTrend(logger, qm)
                    ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sysop/web/lib/plot_data_provider.py", line 297, in __init__
    super().__init__(logger, qm, [], [])
  File "/home/sysop/web/lib/plot_data_provider.py", line 41, in __init__
    self.refresh()
  File "/home/sysop/web/lib/plot_data_provider.py", line 337, in refresh
    qry_data = self.__load_data()
               ^^^^^^^^^^^^^^^^^^
  File "/home/sysop/web/lib/plot_data_provider.py", line 330, in __load_data
    df = df.rolling("30D").mean()
         ^^^^^^^^^^^^^^^^^
  File "/home/sysop/web/venv/lib/python3.11/site-packages/pandas/core/generic.py", line 12577, in rolling
    return Rolling(
           ^^^^^^^^
  File "/home/sysop/web/venv/lib/python3.11/site-packages/pandas/core/window/rolling.py", line 170, in __init__
    self._validate()
  File "/home/sysop/web/venv/lib/python3.11/site-packages/pandas/core/window/rolling.py", line 1877, in _validate
    self._validate_datetimelike_monotonic()
  File "/home/sysop/web/venv/lib/python3.11/site-packages/pandas/core/window/rolling.py", line 1921, in _validate_datetimelike_monotonic
    if self._on.hasnans:
       ^^^^^^^^^^^^^^^^
  File "properties.pyx", line 36, in pandas._libs.properties.CachedProperty.__get__
  File "/home/sysop/web/venv/lib/python3.11/site-packages/pandas/core/indexes/base.py", line 2840, in hasnans
    return bool(self._isnan.any())
                ^^^^^^^^^^^
  File "properties.pyx", line 36, in pandas._libs.properties.CachedProperty.__get__
  File "/home/sysop/web/venv/lib/python3.11/site-packages/pandas/core/indexes/base.py", line 2810, in _isnan
    return isna(self)
           ^^^^^^^^^^
  File "/home/sysop/web/venv/lib/python3.11/site-packages/pandas/core/dtypes/missing.py", line 178, in isna
    return _isna(obj)
           ^^^^^^^^^^
  File "/home/sysop/web/venv/lib/python3.11/site-packages/pandas/core/dtypes/missing.py", line 203, in _isna
    raise NotImplementedError("isna is not defined for MultiIndex")
NotImplementedError: isna is not defined for MultiIndex

not sure how to proceed?  anyone have a suggestion?

Mike- N3BSQ

maserowik, 2024-09-24:
So with the MOJO update of DX Spider path have been changed.  Anyone albe to assist on what need changed on web gui to make it work?

videobuff, 2024-09-28:
I have used the --break-system-packages method, and so far it did not break my server (debian 12). However i got stuck on minifying css. I make a separate issue
PA0ESH

maserowik, 2024-10-16:
Do to get past the message my blocket here is what I did
sudo apt install python3-full libev-dev libevdev2
cd spiderweb
_**python3 -m venv --system-site-packages venv
source venv/bin/activate
I then had the this as a prompt to show I was in my virtual environment   **(venv)** sysop@dx-spider:~/spiderweb$
python -m pip install --upgrade pip
python -m pip install -r requirments.txt
deactivate**_ 
now back to normal prompt sysop@dx-spider:~/spiderweb$
cp /home/sysop/spiderweb/cfg/config.json.template  /home/sysop/spider/web/cfg/config.json
made my changes 
did a _source venv/bin/activate_ and tested with 192.168.55.156:5000
everything worked
then did a deactivate 

I then needed to install bjoern
so I issued a 
_source venv/bin/activate
back to my virtual environment **(venv)** sysop@dx-spider:~/spiderweb$
issues a pip3 install bjoern
deactivate_

This is how I has to set up my siderweb.service
 [Unit]
Description=bjoern instance spiderweb
After=network.target
After=multi-user.target

[Service]
User=sysop
Group=www-data
Type=simple
WorkingDirectory=/home/sysop/spiderweb
Environment="PATH=/home/sysop/spiderweb"
_ExecStart=/home/sysop/spiderweb/venv/bin/python3_ /home/sysop/spiderweb/wsgi.py

[Install]
WantedBy=multi-user.target

i have to use my virtual python3 in order for the service to start.

upon completion I see now that the website is on 192.168.55.156:8080 not sure how it changed from 5000 to 8080 but it appears to be working. 

Not sure if this is the correct way but it is working for me,

Mike- N3bsq

closed 2024-10-16 by maserowik