coulisse/spiderweb #73: No spots shown

videobuff, 2024-10-05:
**Describe the bug**
dxspider is up and running and showing data. Coulisse is up with python3 webapp.py
```
2024-10-05 23:36:48,093 [INFO] (webapp.py) Starting SPIDERWEB
2024-10-05 23:36:48,095 [INFO] (webapp.py) Version:v2.5.4
2024-10-05 23:36:48,095 [INFO] (webapp.py) visit saved on: data/visits.json
2024-10-05 23:36:48,096 [INFO] (cty.py) CTY: start initialization
2024-10-05 23:36:48,096 [INFO] (cty.py) /home/sysop/spider/spiderweb/lib/../data/cty_wt_mod.dat updated (0.0 days), is not necessary to download it
```
The demo build shows the records in the database. So far so good.
However, when i use the release build,  the site is up and running but no spots are shown.
Charts and spots page is showing but no data.
Sometimes i see a warning that states: no data found.
`2024-10-05 23:44:20,777 [WARNING] (webapp.py) no data found`

If the dxcluster mariadb database is totally empty, webapp.py also crashes so i used one dummy record to avoid crashing at startup.

Coulisse is running on a servlet with debian 12. the same where dxspider is running. Bot run with the same user.

**Screenshots**
IHere is a screenshot where you see my telnet windows and coulisse. The data seen in the telnet window is not shown in coulisse.

[image]

Any help is really appreciated becuase it is a fabtastic tool.
Erik, PA0ESH

DocViper666, 2024-10-06:
It only works if the startup order is right.

Database up and running.
Then start DXSpider.

DXSpider has no reconnect to the database and only connects during the start.

You have created the table structure I guess ?
DXSpider has a debug log file. Grep there for the sql errors.

videobuff, 2024-10-06:
Thanks for your remarks.
So i killed all proceesses with systemctl stop mariadb and systemctl stop dxspider. 
Then restarted them; first systemctl start mariadb, en then systemctl start dxspider. 
No errors there. dxspider is working and showing spots in a telnetbox. But the database does not get filled.
I have created the structure as per  instructions. It was not created automatically.
[https://github.com/coulisse/spiderweb/blob/master/docs/DB_ISSUES.md]

However, i have not found the dxspider log yet. Do you know where to locate it ?
I searched for dxspider.log and for *.log but it did not turn up.

DocViper666, 2024-10-06:
The debug logs are created automatically here:
/spider/data/debug/
If there is a problem in inserting data to the mariadb then you should see it here.

videobuff, 2024-10-06:
First problem.
I do not have that directory, so i don't have the debug.log. 
Reading through the manual again to see where i should have set that logging.

videobuff, 2024-10-06:
Reinstalled dxspider using the script by glaucos78. dxspider up and running.
The debug i found in /home/sysop/spider/local_data/debug/2024
So far nothing strange and no sql errors, but also no spots in the database.
In the i discoverd a not which led me ti install libdbd-mysql-perl. Now spots are recorded. Thank you for pushig me in the righ direction.

DocViper666, 2024-10-07:
In my dockerproject you see all the dependencies needed for Dxspider (as far as I know and copy-pasted from the original docker repo):
https://github.com/DocViper666/dxspider-deployment/blob/master/dxspider/Dockerfile

closed 2024-10-08 by videobuff