coulisse/spiderweb #74: Difference to nodes/users

videobuff, 2024-10-10:
I finally managed the installation.
OS server is Debian 12
Database is Mariadb
proxy server is nginx.

[](adres is https://dxcluster.pa0esh.nl)

However, if i give the command `who `at dxspider and look at the 2nd page of the coulisse site at the nodes and users, there is a differance.

See the pictures. Top is coulisse, bottom is telnet.

[image]
[image]

Can anyone explain.

73's Erik

maserowik, 2024-10-16:
were you albe to solve this issue?

Mike- N3BSQ

coulisse, 2024-11-29:
Connecting to your website now I see:

[image]

If you do a `who` what do you see?

coulisse, 2024-12-07:
Since there are not any news, I close this issue.
If  further help is needed, please reopen it.

closed 2024-12-07 by coulisse