coulisse/spiderweb #78: spiderweb scripts unable to host the website

Minitajfun, 2024-12-23:
**Describe the bug**
When starting spiderweb, either via `wsgi.py` or `webapp.py`, the program fails to host the actual website despite running with no errors reported.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `wsgi.py` or `webapp.py`
2. Wait for a bit
3. Website is not hosted, no related errors or warnings in the log 

**Expected behavior**
Program should host a website

**Server:**
 - OS: Ubuntu 24.04.1 LTS
 - Program Version: 2.7.5
 - Python Version: 3.12.3
 - Python Pkg Versions: same as in [a5a96646f4e5bbec12105b957d2e39bc2b01c6ab](https://github.com/coulisse/spiderweb/blob/e28f7cdf7098661539e67f252138ff1219c1c3e6/requirements.txt)

**Additional context**
While hosting the website does not work when running either of two scripts, running `flask run` successfully hosts a website, which is weird because, to my understanding, `flask run` should behave the same exact way as `Flask.run()`.

**Log file**
[webapp.log](https://github.com/user-attachments/files/18232433/webapp.log)

Minitajfun, 2024-12-27:
After checking the log today, I was greeted by this message:
```
[ERROR] (dxtelnet.py) could not connect to telnet dxspider host: Separator is not found, and chunk exceed the limit
```

After a bit of investigation, the issue turned out to be a `WAIT_FOR` variable in the `who()` function of `dxtelnet.py`.
After logging in to telnet, the program will read until it finds "dxspider >" string. However, and I might not be 100% right here, if you log in using a non-anonymous user, a different prompt will be displayed, so for example instead of
```
[MY_CALLSIGN] de IU1BOW-2 27-Dec-2024 1918Z dxspider >
```
you will get
```
[MY_CALLSIGN]>
```
which will cause the program to hang.

Changing this variable to my callsign solved my problem, but I'm leaving this issue open as I believe it should be properly resolved.

coulisse, 2024-12-28:
Hello,
in order to solve this issue, please change in the cfg/webapp_log_config.ini,  the leves for all the keyes, from info to debug.

level=INFO --> level=DEBUG

Then restart the application and give me the log.

Moreover, if it is possible, could I make some test with your cluster? 
In this case give me your credential using my mail corrado.gerbaldo@gmail.com

Best regards
Corrado

coulisse, 2025-01-09:
Hello, have you any news?

Minitajfun, 2025-01-12:
Hi,
Sorry for the late response. I was sure I sent you the log and credentials already, but I guess my memory resembles Swiss cheese.

As I write this, I'm sending you the cluster credentials in the mail (along with the log file for convenience). 

And here's a copy of the log: [webapp.log](https://github.com/user-attachments/files/18390098/webapp.log)

Minitajfun, 2025-05-04:
Hi,
Looks like the update solved the issue.
Thanks for the help!

Minitajfun, 2025-05-04:
Alright, turns out I should've given it a deeper look.

The website will now load correctly, but the "telnet users & nodes online" section from *Charts & stats* no longer seems to list any information.

[Here's a debug log](https://github.com/user-attachments/files/20027155/webapp_debug.log)

coulisse, 2025-05-05:
Which version and build of DxSpider are you using?
More over could you give me the output of "who" command on your DxSpider ?

Minitajfun, 2025-05-07:
DxSpider Version: `DXSpider v1.57 build 633`
"who" output:
```
who
  Callsign Type      State    Started           Name     Ave RTT Link
   CX2SA-8 NODE DXSP normal    7-May-2025 0449Z             0.57 201.217.129.150
  EA4FIT-2 NODE DXSP normal    7-May-2025 0450Z             0.16 79.117.32.117
    GB7VAX NODE DXSP normal    7-May-2025 0450Z             0.14 86.173.57.45
  IK4PKL-6 NODE DXSP normal    7-May-2025 0444Z             0.17 81.57.22.115
  IK7IMO-6 NODE DXSP normal    7-May-2025 0445Z             0.17 37.101.4.229
    IZ0FKE NODE DXSP normal    7-May-2025 1305Z            17.19 94.34.69.93
  IZ5ILU-6 NODE DXSP normal    7-May-2025 0440Z             0.16 217.198.133.86
  JH1RFM-9 NODE DXSP normal    7-May-2025 0440Z             0.41 126.89.253.58
  KB3CMT-2 NODE DXSP normal    7-May-2025 0443Z             0.19 100.6.87.173
    SQ3MZM USER LOCL prompt    7-May-2025 1415Z Tom              127.0.0.1
  SQ3MZM-3 NODE DXSP indiffer  7-May-2025 0439Z Tom
  WB3FFV-2 NODE DXSP normal    7-May-2025 0440Z             0.48 204.9.93.14
```

coulisse, 2025-05-10:
Hello
I tried to connect to you cluster and I see the version and the connection:

[image]

This is the configuration  I used:
    "telnet": {
        "telnet_host": "sq3mzm.ddns.net",
        "telnet_port": "7300",
        "telnet_user": "IU1BOW",
        "telnet_password": ""
    },

Minitajfun, 2025-05-21:
<details>
<summary>Sorry for the late reply, github decided that I do not need notifications for this issue</summary>
https://github.com/user-attachments/assets/8c5592d1-7de7-4c6c-9247-54f53a58cf24
</details>

Anyways, as mentioned [here](https://github.com/coulisse/spiderweb/issues/78#issuecomment-2563990684), if you log in with your callsign to the telnet you will see this prompt
```
IU1BOW de SQ3MZM-3 21-May-2025 1828Z dxspider >
```
However, when I log in with SQ3MZM, I get this prompt
```
SQ3MZM-3>
```

Function `who()` from `dxtelnet.py` reads data until it finds "dxspider >", but because my callsign's prompt does not end with it, it causes the function to read until failure.

I honestly do not know what causes the prompt to be different. I definitely did not modify it, and even if I wanted to, I wouldn't know where to begin.
I guess at this point it's more of a DXSpider issue than Webspider one, but any help on fixing this difference would be appreciated.