coulisse/spiderweb #79: Website unique visits

s50u, 2024-12-24:
With us "Website unique visits" always shows 1. Of course FLASK calls the local user 127.0.0.1 and according to /data/visits.json the counter is working.

{"127.0.0.1": 206}

**mod_remoteip** module is loaded and we put in Apache the following command which should help but doesn't work:

RemoteIPHeader X-Forwarded-For
    RemoteIPTrustedProxy 127.0.0.1

Server specifications:

Linux **Ubuntu 22.04.4**
**Apache/2.4.52** (Ubuntu)
Mysql  Ver 15.1 Distrib 10.6.18-MariaDB
PHP 8.1.2-1ubuntu2.20 (cli)

Has anyone dealt with this problem to get client IP address count correctly? 😕

Thanks, Dan

coulisse, 2024-12-28:
I could not to reproduce this issue
Could you change the user_ip istruction in the function visitor_count (in webapp.py) in this way:

```
    #user_ip =request.environ.get('HTTP_X_REAL_IP', request.remote_addr)   
    user_ip = request.environ.get('HTTP_X_FORWARDED_FOR', request.remote_addr)  
```

Tell me if this change anythings

s50u, 2024-12-28:
> user_ip = request.environ.get('HTTP_X_FORWARDED_FOR', request.remote_addr)  

Yes I think this solution works for me at least. Thank you for your support.

HNY 2025 Dan S50U

coulisse, 2024-12-28:
Good!
Since I couldn't make a test with your environment, can change the instruction with this?

`    user_ip = request.environ.get('HTTP_X_FORWARDED_FOR') or request.environ.get('HTTP_X_REAL_IP') or request.remote_addr`

**Explanation**:

_user_ip = request.environ.get('HTTP_X_FORWARDED_FOR')_

This line attempts to get the IP address from the HTTP header X-Forwarded-For. This header is often used in proxy environments or behind a load balancer, where the user's original IP address may be hidden.

_or request.environ.get('HTTP_X_REAL_IP')_

If the X-Forwarded-For header is not present or does not contain the correct IP address, the code tries to get the IP address from the X-Real-IP header. This header can be used in some cases to get the user's real IP address.

_or request.remote_addr_

If neither of the previous headers provides the IP address, the code uses the remote_addr attribute of the request object. This attribute generally contains the IP address of the client that made the request.

s50u, 2024-12-28:
> Good! Since I couldn't make a test with your environment, can you change the instruction with this?

This is how we have arranged it on Ubuntu 22.04.4 server. We've changed the line in **webapp.py**

from

```
user_ip =request.environ.get('HTTP_X_REAL_IP', request.remote_addr)
```

to

```
user_ip = request.environ.get('HTTP_X_FORWARDED_FOR') or
request.environ.get('HTTP_X_REAL_IP') or
request.remote_addr
```

We are using a virtual hosts for spiderweb with directives in **/etc/apache2/sites-available/spiderweb.conf** as set out below. We use a proxy environment as we convert port 443 to 8181.

```
    <VirtualHost *:443>
    ServerName s50clx.infrax.si
    ServerAdmin webmaster@localhost

    LoadModule proxy_module /usr/lib/apache2/modules/mod_proxy.so
    LoadModule proxy_http_module /usr/lib/apache2/modules/mod_proxy_http.so
    LoadModule remoteip_module /usr/lib/apache2/modules/mod_remoteip.so

    RemoteIPInternalProxy 127.0.0.1 10.1.1.1
    RemoteIPHeader X-Forwarded-for

    ProxyRequests Off
    ProxyPass / http://127.0.0.1:8181/
    ProxyPassReverse / http://127.0.0.1:8181/

    ErrorLog ${APACHE_LOG_DIR}/error.log
    LogLevel warn
    CustomLog ${APACHE_LOG_DIR}/access.log combined

    SSLEngine On

    Include /etc/letsencrypt/options-ssl-apache.conf
    SSLCertificateFile /etc/letsencrypt/live/s50clx.infrax.si/fullchain.pem
    SSLCertificateKeyFile /etc/letsencrypt/live/s50clx.infrax.si/privkey.pem
</VirtualHost>

```
HNY 2025 Dan S50U

commit 6fe4cc23 (2024-12-28): Fix #79
closed 2024-12-29 by coulisse