coulisse/spiderweb #80: Plots & Stats

maserowik, 2024-12-26:
so on my plots&stats I have the following showing up

[image]

but doing a who on my cluster I have this 
[image]

notice on the dxcluster who command I have 9m2pju-2 showing up but that call is missing on the plots& stats web page.

Thanks
Mike- N3BSQ

s50u, 2024-12-26:
> notice on the dxcluster who and show/conn command I have 9m2pju-2 showing up but that call is missing on the plots& stats web page.

In webapp.py change the parameter **TIMER_WHO = 2 * 60** and you will have fresher user data. The module will connect to the cluster every two minutes and retrieve users. Currently it is ten minutes by default.

HNY 2025 Dan S50U

maserowik, 2024-12-26:
> In webapp.py change the parameter **TIMER_WHO = 2 * 60** and you will have fresher user data. The module will connect to the cluster every two minutes and retrieve users. Currently it is ten minutes [...]

In the config.sh I have the spot set to refresh 1000    

[image]

s50u, 2024-12-26:
> In the config.sh I have the spot set to refresh 1000

Mike
The refresh time of the spots has no relation to the user&node list. A security enhancement for the telnet connection to retrieve the WHO list is written in /spiderweb/docs. Check issues #69 acording to link below.

Date: 22/12/2024
Release: v2.5.7
- Improved telnet connection - Issue [#69](https://github.com/coulisse/spiderweb/issues/69)

maserowik, 2024-12-26:
I am currently  on 2.5.7.  According to the release docs this issue is solved?   Now I am confused.  The is not a user it is a dx node.Michael A. SerowikSent from my iPhone

s50u, 2024-12-27:
> I am currently  on 2.5.7.  According to the release docs this issue is solved?   Now I am confused.  The is not a user it is a dx node.

I found the problem.
Very the first line of the dxspider WHO command is ignored. It will need finishing by the author in the next release.

HNY 2025, Dan S50U

coulisse, 2024-12-27:
Thanks for your debug, I'll work on it

coulisse, 2024-12-27:
Hello
I committed on "development" brach a new version, in order to fix this issue? Could anyone make any further tests?

closed 2024-12-29 by coulisse

maserowik, 2024-12-29:
Thanks I was away for  Christmas and just did the update and it is working correctly. 
 
[image]

commit 12be60c3 (2024-12-27): Fix #80