coulisse/spiderweb #85: Version on frontpage not changed

videobuff, 2025-01-07:
Just updated from github, and i see that version has not changed on the webpage.
The version.txt in the cfg directory however, contains the correct version (v2.5.7.4)
the file _base.html also contains the correct version
<span id="version">v2.5.7.4</span>
I use both Chrome and Safari and both are the same old version number 2.5.4
OS is Seqioa 15.2

[image]
 
Emptied all the browser caches multiple times but no change.
Erik

s50u, 2025-01-07:
Erik

I don't want to be too clever but surely you've done the spiderweb app restart with **sudo systemctl restart spiderweb**?
Unfortunately I also see that you are still on v2.5.4 on FF 133.0.3.

Dan

videobuff, 2025-01-07:
Dan, thanks for the hint. I hadn't restarted it. Now that i did i can no longer reach the site.
I get a bad gateway fresult. 
Investigating what i forgot.
73's
Erik

coulisse, 2025-04-20:
Since no news, I close this issue

closed 2025-04-20 by coulisse