coulisse/spiderweb #86: Update version

wifi75, 2025-01-10:
Hi, what is the correct configuration to update to the latest version without losing your configuration?

s50u, 2025-01-12:
The basic approach would be:

- **save your spiderweb directory in archive directory** e.g. spiderweb_old
- delete spiderweb folder
- install needed modules using requirements.txt file
- git clone the new version in spiderweb directory
- copy the cfg/config.json from spiderweb_old to the spiderweb directory

In case of problems reimburse your project from your spiderweb_old archive directory.

Good luck, Dan S50U

wifi75, 2025-01-12:
> The basic approach would be:
> 
> * **save your spiderweb directory in archive directory** e.g. spiderweb_old
> * delete spiderweb folder
> * install needed (some new ones might be needed) modules using requirements.txt file
> * git clone the new version in spiderweb directory
> * copy the cfg/config.json from spiderweb_old to the spiderweb directory
> 
> In case of problems reimburse your project from your spiderweb_old archive directory.
> 
> Good luck, Dan S50U

yes, but not working see bwlow:

[image]

s50u, 2025-01-12:
> > The basic approach would be:
> > * install needed (some new ones might be needed) modules using requirements.txt file

> yes, but not working see bwlow:

Among others you also need to install telnetlib3 which you don't have installed with **pip install -r requirements.txt**

GL Dan S50U

wifi75, 2025-01-12:
how I can install telnetlib3 ?

wifi75, 2025-01-12:
I tried to install but it is already installed...

[image]

wifi75, 2025-01-14:
I still can't get it to start, it always gives me the same error as in the first post.

coulisse, 2025-01-15:
The problem is that you are trying to install telnetlib3 on your user installation, but is it possible that spiderweb3 is runnig with another user?

coulisse, 2025-04-20:
Since no news, I close this issue

closed 2025-04-20 by coulisse