coulisse/spiderweb #89: Edit page

wifi75, 2025-01-31:
hello, how I can edit follow red arrow line with my correct information?

[image]

maserowik, 2025-01-31:
Configuration
In the path spiderweb/cfg/ rename config.json.template in config.json:

foo@bar:~$ mv config.json.template config.json
then edit it and set the user and password of your database, the menu items, and other stuffs (callsign, mail address...). There is also a specific parameter, named "enable_cq_filter" used to enable the CQ Zone filtering.

Othewhise, if you prefer, you could use a utility to edit your configuration and menu. Go in "script" folder and run ./config.sh

foo@bar:~$ cd scripts
foo@bar:~$ ./config.sh

*** DxSpider configuration ***
Configuration file loaded from: ../cfg/config.json

   h:  help
   vc: view config.
   ec: edit config.
   vm: view menu
   em: edit menu
   s:  save
   t:  load config. from template

   x:  exit

Make your choice:

wifi75, 2025-02-03:
> Configuration In the path spiderweb/cfg/ rename config.json.template in config.json:
>
> foo@bar:~$ mv config.json.template config.json then edit it and set the user and password of your database, the [...]

hello maybe i didn't explain myself well, my webcluster works fine, only that i would like to show on the main page the correct link to connect via telenet which is webcluster.iu3cyv.eu on port 7300. this is the correct address that i would like to show on the webcluster home page. how can i do it?
a note of clarification: my webcluster and dxspider run on the same server so the database reads it via localhost. i don't want to make mysql public.

maserowik, 2025-02-03:
This email has been checked for viruses by Avast antivirus software.
www.avast.com

wifi75, 2025-02-08:
I still don't understand...

coulisse, 2025-04-20:
In order to change the telnet access in the header of the website you have to change the data in **config.json** (in cfg folder)

```
    "telnet": {
        "telnet_host": "mysite",
        "telnet_port": "7300",
        "telnet_user": "myuser",
        "telnet_password": ""
    },
```

Pay attention that this is also used in the "Charts & Stats" page, in order to show the on line users.

closed 2025-04-20 by coulisse